dresden-elektronik/deconz-rest-plugin #1634: TRÅDFRI Wireless dimmer: Adding works, controlling anything does not

rb2k, 2019-06-26:
I have successfully added the dimmer via the PWA webinterface.
I assigned it a dimmable lamp.

When I turn it, the lamp doesn't do anything, but I see these in the logs:

```
11:07:56:931 button 2002 Move up (with on/off)
11:07:56:931 button 2002 Move up (with on/off), discard too fast event (dt = 208)
11:07:56:963 button 2002 Move up (with on/off)
11:07:56:963 button 2002 Move up (with on/off), discard too fast event (dt = 240)
11:07:57:027 no button handler for: TRADFRI wireless dimmer ep: 0x01 cl: 0x0008 cmd: 0x07 pl[0]: 000
11:07:57:059 no button handler for: TRADFRI wireless dimmer ep: 0x01 cl: 0x0008 cmd: 0x07 pl[0]: 000
11:07:57:091 no button handler for: TRADFRI wireless dimmer ep: 0x01 cl: 0x0008 cmd: 0x04 pl[0]: 0FF
11:07:57:507 button 2002 Move up (with on/off)
11:07:57:539 button 2002 Move up (with on/off)
11:07:57:539 button 2002 Move up (with on/off), discard too fast event (dt = 32)
11:07:57:571 no button handler for: TRADFRI wireless dimmer ep: 0x01 cl: 0x0008 cmd: 0x07 pl[0]: 000
11:07:57:603 no button handler for: TRADFRI wireless dimmer ep: 0x01 cl: 0x0008 cmd: 0x07 pl[0]: 000
11:07:57:635 button 2002 Move up (with on/off)
11:07:57:635 button 2002 Move up (with on/off), discard too fast event (dt = 128)
11:07:57:667 no button handler for: TRADFRI wireless dimmer ep: 0x01 cl: 0x0008 cmd: 0x07 pl[0]: 000
11:07:57:826 button 4002 Move to level 0 (with on/off)
11:07:57:859 button 4002 Move to level 0 (with on/off)
11:07:57:859 button 4002 Move to level 0 (with on/off), discard too fast event (dt = 32)
```

Is the "no button handler for" an issue?

Philje123, 2019-06-28:
I have a few of these. They are a little tricky sometimes.

Try removing the bulb again and hitting save, turning the dimmer a few times, then add the bulb again and hit save and do the same.

I've found it takes a couple of tries sometimes for it to actually save the assigned bulbs.

MickL, 2019-06-28:
I have the exact same problem! Im using latest deconz version with a Raspbee:

First of I have trouble connecting:

- I press the button 4 times and the next button does not activate
- After 20 tries when im about to give up it suddenly it works ?!
- Then it tells me to turn the dimmer clockwise but the finish button never activates
- On my third try of the whole process the clockwise page activated the finish button directly without me ever touching the dimmer

Now it is connected and i have a Tradfri lightbulp in its group, but turning the dimmer does nothing!

MickL, 2019-06-28:
Okay i tried to connect the dimmer manually to the light. It seems like the dimmer and a tradfri light with different color temperatures are not compatible. The manual says so too. 

Anyway would be nice if deconz/phoscon would add support while we can change in the ui if we want to toggle brightness or temperature.

wucherpfennig, 2019-07-12:
I do not understand:

* __... i tried to connect the dimmer manually__
  I am able to add the dimmers without any problems. they just do not dim anything :-/

How can we help to find a solution?

wucherpfennig, 2019-07-12:
Update

According to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/364#issuecomment-360856214 if found a "solution" that worked...

I do not know how exactly but I am now able to use my dimmers...

* Remove the dimmer
* Add the dimmer via the webui. Choose "others"
* During the dialog (3min) reset the dimmer (by pressing 4 times the connect button)
* Wait till connected. (__Press once again the connect button__)
* Try to select the dimmer in the ui (to assign lights to it...). If it does not work start again
* if able to select the dimmer. add 1 light. __before saving press once again the connect button__
* wait and test
* if not working: remove the light and try again (__press once again the connect button before or while or after saving)
* add more lights if working

honestly: I am happy the dimmers work now but I can really not tell why... the only thing that made them work was the "press the connect button once again" I assume

feedback appreciated

rb2k, 2019-07-13:
Worked for me as well!

rb2k, 2019-07-22:
I guess this is still pretty messy, but I don't see anyone doing anything about it... Guess I'll close it.

MickL, 2019-07-23:
Why closing when this issue is still present? 😳

rb2k, 2019-07-23:
I can reopen it, but so far there wasn’t any movement :)

MickL, 2019-07-23:
But even tho there is no movement the issue still persists or am I wrong? Me and probably a lot of others are hoping for movement here. As a workaround I had to configure the dimmer manually with node-red which was not easy.

bfayers, 2019-07-27:
> 
> 
> Update
> 
> According to [#364 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/364#issuecomment-360856214) if found a "solution" that worked...
> 
> I do not know how exactly but I am now able to use my dimmers...
> 
>     * Remove the dimmer
> 
>     * Add the dimmer via the webui. Choose "others"
> 
>     * During the dialog (3min) reset the dimmer (by pressing 4 times the connect button)
> 
>     * Wait till connected. (**Press once again the connect button**)
> 
>     * Try to select the dimmer in the ui (to assign lights to it...). If it does not work start again
> 
>     * if able to select the dimmer. add 1 light. **before saving press once again the connect button**
> 
>     * wait and test
> 
>     * if not working: remove the light and try again (__press once again the connect button before or while or after saving)
> 
>     * add more lights if working
> 
> 
> honestly: I am happy the dimmers work now but I can really not tell why... the only thing that made them work was the "press the connect button once again" I assume
> 
> feedback appreciated

> 
> 
> Update
> 
> According to [#364 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/364#issuecomment-360856214) if found a "solution" that worked...
> 
> I do not know how exactly but I am now able to use my dimmers...
> 
>     * Remove the dimmer
> 
>     * Add the dimmer via the webui. Choose "others"
> 
>     * During the dialog (3min) reset the dimmer (by pressing 4 times the connect button)
> 
>     * Wait till connected. (**Press once again the connect button**)
> 
>     * Try to select the dimmer in the ui (to assign lights to it...). If it does not work start again
> 
>     * if able to select the dimmer. add 1 light. **before saving press once again the connect button**
> 
>     * wait and test
> 
>     * if not working: remove the light and try again (__press once again the connect button before or while or after saving)
> 
>     * add more lights if working
> 
> 
> honestly: I am happy the dimmers work now but I can really not tell why... the only thing that made them work was the "press the connect button once again" I assume
> 
> feedback appreciated

Thanks! This is the only way I've got my Tradfri dimmer to work in deCONZ.

closed 2019-12-01 by stale[bot]

airbone42, 2020-06-27:
Sorry for bringing this up again. The pairing works very fine for me, but after several days the dimmers stop sending commands. Had this now with 4 different dimmers and the time how long they work is always different. When they stop sending data the only way to re-connect them is by a reset and re-pairing.