dresden-elektronik/deconz-rest-plugin #1635: Conbee 2 and IKEA Tradfri gateway simultaneously

wucherpfennig, 2019-06-27:
Is it possible to have lights / switches connected to the Tradfri gateway and the Conbee 2 simultaneously?

Either I am capable of connecting to Tradfri gateway or to Conbee 2 because I need to "reset" the devices each time but I was not able to establish a working connection to both.

The reason I want to keep the connection with the gateway is simple: updates...

Any suggestions appreciated.

BR wucherpfennig

ebaauw, 2019-06-27:
In theory: yes, in practice: probably not.  See #21.

Even if you get this running, there will be issues: both the Trådfri hub and deCONZ gateway will want to configure the lights for attribute reporting, causing to other gateway no longer to receive updates of the light state.

wucherpfennig, 2019-06-27:
So, best thing is to use a combination of both... (Probably conbee for sensors etc)

What happens to accessibility / address of a unit which I remove temporarily from conbee and connect to the IKEA gateway in order to update the device? Eg. The configuration in a third party system like home assistant

ebaauw, 2019-06-27:
Best not remove the device from the deCONZ network - just reset it.  DeCONZ will no longer see it and mark it unreachable, but the resources (and groups, rules, schedules, etc) remain untouched.  After upgrade, reset the device again, and re-pair it to deCONZ, that will re-connect it to the existing resources.

closed 2019-11-01 by stale[bot]

ln-12, 2019-11-03:
Hey guys, I somehow got it working that [this lamp](https://www.ikea.com/de/de/p/tradfri-led-leuchtmittel-e14-400-lm-kabellos-dimmbar-warmes-tageslicht-kerzenfoermig-opalweiss-60365271/) is both connected to my ConBee and [this remote](https://www.ikea.com/de/de/p/tradfri-kabelloser-dimmer-weiss-70408595/). That means I have full control over the lamp including dimming in Home Assistant and can dimm it with my remote at the same time.

Unfortunately, we trying to set this scenario up with [this lamp](https://www.ikea.com/de/de/p/tradfri-led-leuchtmittel-e14-600-lm-kabellos-dimmbar-weissspektrum-opalweiss-80408585/) and [this remote](https://www.ikea.com/de/de/p/tradfri-fernbedienung-30443124/), I can only pair either the remote OR the ConBee to my lamp. Do you have an explanation why this works in the first case but not for my second setup?

ebaauw, 2019-11-03:
The remote and the lamp need to be in the ZigBee network that's created by the ConBee.  I'm not sure if that's possible through touchlinking, but you would have to pair the remote to the ConBee first.  Better use deCONZ, and pair both the light and the remote to it.  Then, add the light to the group that's controlled by the remote (see the ZHASwitch sensor's `config.group`).

ln-12, 2019-11-03:
Thanks for the fast response! After following your suggested steps I can now control the light via Home Assistant and via the remote. I can turn it on/off and adjust the brightness, even if the Raspberry is not running. The only thing that does not work when setting it up this way is changing the color temperature. Maybe that's some kind of software configuration problem? I could try setting up an automation rule in HA, but it would be great, if it worked out of the box.

Smanar, 2019-11-03:
Yep it's normal, on/off and britghtness are native zigbee command , but for the color the remote use scene command (next scene / previous scene) so for that you need a "manfacturer inclusion" that configure theses scenes.

ln-12, 2019-11-03:
Ok, so is there a way to get this working in the future? Or does IKEA have to fix it?

The automation in HA does not really work, becuase I can't even read out the current color_temp attribute in code although I can change the color through the UI slider :(

ebaauw, 2019-11-03:
> The only thing that does not work when setting it up this way is changing the color temperature. Maybe that's some kind of software configuration problem?

IKEA use manufacturer specific commands to change the colour temperature.  These are only recognised by Trådfri colour / colour temperature bulbs in factory-reset configuration.  Once deCONZ configures groups and scenes for these bulbs, the bulbs no longer react to these commands.

Effectively, this means that you cannot use the `<` and `>` buttons standalone, when the remote is connected to the deCONZ ZigBee network.  You can, of course, create deCONZ rules to change the colour / colour temperature when these buttons are pressed.  In fact, as I understand, Phoscon will setup these rules for you.

> Ok, so is there a way to get this working in the future?

In theory, we might be able to figure out these settings and, if technically possible, try for deCONZ to preserve (or restore) them.  You would have to run deCONZ with the REST API plugin disabled, reset and re-pair a Trådfri bulb, and query the _Groups_ and _Scenes_ clusters using the GUI and/or the CLI plugin.  Then manually replay the REST API plugin setup, like adding the bulb to `/groups/0`, and see where the bulb stops reacting to the custom commands.  You might need to sniff the setup of the light by the REST API plugin first.

> Or does IKEA have to fix it?

I have seen other ZigBee remotes that do use standard ZigBee commands to change colour temperature, but I doubt that IKEA would see this as something that needs to be fixed.

ln-12, 2019-11-03:
Thanks for the detailed explanation. Seems like there's quite some stuff to do to get it working perfectly... For now I just use a workaround with hard coded color values in HA instead of dynamically adjusting them.

But nevertheless: thanks for your work on deConz, this tool is just amazing!