dresden-elektronik/deconz-rest-plugin #1636: 3 months working = fixed battery level !

gekomax, 2019-06-27:
Hi,

I'm using Different 8 Xiaomi aqara sensors and 1 tradfri Ikea remote control.
I've been using them from about 3 months
They work correctly but the battery level has not changed during this period of time.
They work vi deconz hassio addon

Any suggestion?
Thanks

Smanar, 2019-06-27:
Hello, wich one deconz version are you using ?

gekomax, 2019-06-27:
Hi, 

My deconz is 3.0 (just updated 1 hour ago).
I use a ConBee usb key installed on raspberry Pi 3B

Thank you

Smanar, 2019-06-28:
Hu ? The last is V2_05_66, where are you seen 3.0 ?
You can find your version on phoscon setting/gateway.

You have too the problem in phoscon ?

gekomax, 2019-06-28:
Sorry, my deconz-hassio-addon is 3.0 (I just noticed there is a new update today)
Deconz firmware is 2.05.66 

Battery level is the same in hassio and phoscom web-app

Smanar, 2019-06-28:
Because it has a battery issue on previous version, but not on this one you have ATM.
After 3 months all my Xiaomi devices are still at 100% too (motion/door/temperature/switch)

gekomax, 2019-06-28:
Thanks! we will see what happens in the next future!

pergolafabio, 2019-07-02:
Ah ok, it's normal the 100% so not an issue?

I have mine like 4 months, all 4 of them are still on 100... When do they update then? When battery is dead?

Smanar, 2019-07-02:
Lol, nope, they update the value every 1h.
I think you will have more than  90% during 1 years for exemple, and the value go from 90% to 10 % in 1 month. It's not something proportional.
On xiaomi the device send a battery voltage value and deconz convert it in a percent value, but on battery the voltage value stay hight for long time, then decrease faster on last days life.

MattL0, 2019-09-30:
here almost all devicers are at 100% after 1 years... I guess i,ll have to change the battry in a couple of months? hard to say...

closed 2020-02-04 by stale[bot]