dresden-elektronik/deconz-rest-plugin #1639: can't install in rapsberry 3 strectch

Klagio, 2019-06-29:
pi@pihole:~ $ sudo apt update
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://phoscon.de/apt/deconz stretch InRelease
Get:3 http://deb.debian.org/debian unstable InRelease [214 kB]
Hit:4 https://repo.mosquitto.org/debian stretch InRelease
Get:5 http://download.zerotier.com/debian/stretch stretch InRelease [20.0 kB]
Err:3 http://deb.debian.org/debian unstable InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7638D0442B90D010 NO_PUBKEY 04EE7237B7D453EC
Hit:6 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Reading package lists... Done
W: GPG error: http://deb.debian.org/debian unstable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7638D0442B90D010 NO_PUBKEY 04EE7237B7D453EC
E: The repository 'http://deb.debian.org/debian unstable InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
pi@pihole:~ $ sudo apt install deconz
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package deconz
pi@pihole:~ $

Klagio, 2019-06-29:
Nothing, I can't install deCONZ on a raspberrry pi3 following the documentation

manup, 2019-06-29:
>  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 7638D0442B90D010 NO_PUBKEY 04EE7237B7D453EC 

Doesn't look deCONZ specific. It seems that your system is is configured for unstable Debian repositories and is missing pubic keys to verify these packages.

You need to add these as described here (replace with the keys from the error log):

https://ccm.net/faq/809-debian-apt-get-no-pubkey-gpg-error

closed 2019-11-03 by stale[bot]