dresden-elektronik/deconz-rest-plugin #1640: WARN Cannot submit Set TR Deq Ptr

freenetwork, 2019-06-29:
Before the update, there was a reboot cycle, put the latest firmware through windows and it became better. However, dmesg writes the following

[image]

```
[ 2659.320808] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[ 2659.320812] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[ 2659.322618] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[ 2659.322623] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[ 2659.322721] cdc_acm 1-2.1:1.0: failed to set dtr/rts
[ 2659.324708] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[ 2659.324713] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[ 2659.552653] usb 1-2.1: new full-speed USB device number 102 using xhci_hcd
[ 2659.681482] usb 1-2.1: New USB device found, idVendor=1cf1, idProduct=0030
[ 2659.681488] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2659.681492] usb 1-2.1: Product: ConBee II
[ 2659.681496] usb 1-2.1: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[ 2659.681500] usb 1-2.1: SerialNumber: DE1964017
[ 2659.683329] cdc_acm 1-2.1:1.0: ttyACM0: USB ACM device
[ 2760.423132] docker0: port 3(veth1b23855) entered blocking state
[ 2760.423136] docker0: port 3(veth1b23855) entered disabled state
[ 2760.423601] device veth1b23855 entered promiscuous mode
[ 2760.428093] IPv6: ADDRCONF(NETDEV_UP): veth1b23855: link is not ready
[ 2760.999886] eth0: renamed from vetha2d30ba
[ 2761.011090] IPv6: ADDRCONF(NETDEV_CHANGE): veth1b23855: link becomes ready
[ 2761.011158] docker0: port 3(veth1b23855) entered blocking state
[ 2761.011161] docker0: port 3(veth1b23855) entered forwarding state
[ 2808.751005] docker0: port 3(veth1b23855) entered disabled state
[ 2808.753936] vetha2d30ba: renamed from eth0
[ 2808.807883] docker0: port 3(veth1b23855) entered disabled state
[ 2808.822180] device veth1b23855 left promiscuous mode
[ 2808.822190] docker0: port 3(veth1b23855) entered disabled state
[ 2980.942576] udevd[7]: starting version 3.2.7
[ 2980.952809] udevd[8]: starting eudev-3.2.7
```

manup, 2019-06-29:
Haven't seen that one yet, perhaps the following is related:

https://bugzilla.kernel.org/show_bug.cgi?id=202541

Which machine do you have?

manup, 2019-06-29:
And which Linux kernel?

    uname -a

closed 2019-11-03 by stale[bot]