dresden-elektronik/deconz-rest-plugin #1641: Conbee II - make firmware 0x26420700 available

dbdbdb, 2019-06-30:
I have bought a conbee II with firmware 0x26420700. I've developed an application that speaks the serial protocol and it works fine.

There are new firmwares I can install but I can't find the 0x26420700 that is currently installed in my stick. I want to be able to downgrade again to make sure that the application works with all versions of the firmware.

Would it be possible for you to put the 0x26420700 somewhere for download?

Kind regards,
Dennis

Kane610, 2019-06-30:
So many improvements to fw so far and it's quite new. Wouldn't it be better to require the latest available right now? Makes life easier for you

dbdbdb, 2019-06-30:
I'm sure the latest version is much better but it's still nice if the code work with whatever version is in the stick. I use an old conbee myself and I bought a new conbee II just to make sure the code work with it as well.

I already have an easy life :-)

So far the differences I've seen are minor. Like the Device State response has a frame length of 7 instead of the documented 8. Plus the watchdog stuff that is new of course.

manup, 2019-07-03:
>Would it be possible for you to put the 0x26420700 somewhere for download?

It is online now, but keep in mind this versions has some real issues and updating for users is highly recommended.

https://www.dresden-elektronik.de/rpi/deconz-firmware/

dbdbdb, 2019-07-06:
Great. Thanks a lot!

closed 2019-07-06 by dbdbdb