dresden-elektronik/deconz-rest-plugin #1642: What are reliable US smart plugs that work as  routers for Xiaomi devices?

oleggold, 2019-06-30:
Please recommend a reliable US smart plug that work as a router for Xiaomi devices in Deconz world.
So far I tried without success the following two smart plugs:
1. SYLVANIA SMART +, model 72922-A (72922 without A isn't available anymore) https://www.amazon.com/gp/product/B01M6UM8QD/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1
2. TRÅDFRI Wireless control outlet. https://www.ikea.com/us/en/catalog/products/30356169/#BVContainer

They both connect to Deconz without any issues, but Xiaomi Motion Sensor would not connect to either.

Thank you.

thowi, 2019-07-05:
I don't have an answer for you (I'm based in EU), but I found that for Xiaomi sensors it's important that they are very very close to the router when pairing, so they use it as a parent.
Maybe you've tried this already, but otherwise worth a shot.

Smanar, 2019-07-05:
I m interested too by this question. For US but EU too.
ATM I m using only bulb as router.

oleggold, 2019-07-09:
I also loaded a router firmware on CC2531 stick and connected it to deconz. It showed up appropriately as a router, but Xiaomi sensors will not use it in the mesh. My deconz mesh picture (see attached) also looks strange -- there are no connected lines to some of the sensors, but they are working properly.
Is this normal?
[image]

thowi, 2019-07-09:
I don't have a CC2531, @oleggold, but I've got a couple of other routers (smart plugs), and Xiaomi sensors still tend to use the Conbee coordinator as the parent, even though the routers are much closer to the sensor. Not sure why. Maybe the Conbee answers faster or something.
It hasn't been a problem for me yet, though, as the connection directly to the Conbee coordinator seems reliable.

Smanar, 2019-07-09:
I think yes @oleggold, another person use a CC2530 (with suceess) with same result than you https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1428

oleggold, 2019-07-09:
Thanks all for your comments.
I need to use a router, so I can get the Xiaomi sensors to work far away from the Conbee II.
So, far when I place Xiaomi sensor in the desired location (far from Conbee II, but fairly close to a working router, I get it disconnected from Conbee II, and it never connects to a router.
Maybe someone from dresden-elektronik is willing to help?

pergolafabio, 2019-07-09:
i am using osrams, my xiaomi pir are are connected to them just fine

Smanar, 2019-07-09:
I think you are lucky @pergolafabio > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1595#issuecomment-502462107

@oleggold you can't deplace a Xiaomi sensor, this kind of device don't change his parent, if it loose it, il will leave the network, and don't reconnect.
The correct methos is re-pairing it, close the router you want, it let it in place for some hours (during the first hours, I have ever see a Xioami device changing his parent).
But the correct method is pairing device at their definitive place, if the device prefer the direct connexion, it have a reason.

And from I have understand, if you haven't "green lines" it mean you are using the CC253X router, so you are fine, nope ?

pergolafabio, 2019-07-09:
so i am lucky because its stays connected to the osram? so if i take out the osram? it doesnt reconnect/find to another router? is that the issue with osram?

Smanar, 2019-07-09:
Nope ^^, if it works, don't touch something.
If one day, you have problem with your device, look at this plug. You are not alone with an osram plug and no problems.

oleggold, 2019-07-09:
> I think yes @oleggold, another person use a CC2530 (with success) with same result than you #1428 it works, but "green lines" are invisibles.

Thanks for pointing me in the right direction.

oleggold, 2019-07-11:
I now can confirm that as reported in #1428, the Xiaomi sensors can connect thru routers, but the connection lines are missing from the Map. The end devices that do not have any connected lines are all working properly and are connected thru a router (see below).
Maybe deconz developers can take a look at this mapping issue.
Thanks again for the help.
[image]

pergolafabio, 2019-10-21:
any update here? :)
has anyone tried an actual xiaomi smart plug?

anyway, i ditched my osrams, replaced by ikea smart plug, those ikea smart plugs are MUCH better then the osrams, all green instead of orange/yellow

anyway, started from scratch, and i tried pairing with an ikea , so i took the ikea smart plug, far from conbee, started pairing with xiaomi, yes, first one succeeded with ikea!!! Yey
now , the other 3 , pfff , no luck there, tried it like 20 times, they just wont pair...
if i go closer to conbee; then no issue to pair

so any tips are welcome, as you can see in screenshot, 1 xiaomi is paired with ikea, i want to the other 3 also paired with a ikea plug, because my goal is to move the conbee to another place

https://www.dropbox.com/s/lh1mmoiczxg02ie/mesh.PNG?dl=0

pergolafabio, 2019-11-02:
ok, bought another ikea smart plug, all my xiaomi paired to it, from first time
no idea what was wrong with the other one, maybe i should have deleted it first
anyway, i have now a relaible system, with smart plugs and connected xiaomi pir sensors

stable

Smanar, 2019-11-02:
@pergolafabio So the problem was from the plug ?

pergolafabio, 2019-11-02:
not sure

with the other plug still present, i also changed zigbee channel, tried other things, i was not able to pair the xiaomis
now with the new plug, same procedure, all 3 paired instant with it!
with the other plug, the only thing i didnt test, was to delete the plug from the mesh and start clean.... maybe that was the trick

now my mesh is stable for my end devices! working really good
screenshot : https://www.dropbox.com/s/vwu91a37losuex0/mesh-final.PNG?dl=0

Smanar, 2019-11-02:
On your screenshoot you have already loose 2 devices ^^

pergolafabio, 2019-11-02:
2 loose? no just one :-)  , that one is without battery , not using it at the moment, so its normal :-)

oleggold, 2019-11-06:
> ok, bought another ikea smart plug, all my xiaomi paired to it, from first time
> no idea what was wrong with the other one, maybe i should have deleted it first
> anyway, i have now a relaible system, with smart plugs and connected xiaomi pir sensors
> 
> stable

Are you in US?

pergolafabio, 2019-11-06:
no belgium, but the plug is the same? us should be the same as eu version? is just the adapter thats different

oleggold, 2019-11-06:
In my OP I mentioned that I already tried 
TRÅDFRI Wireless control outlet. https://www.ikea.com/us/en/catalog/products/30356169/#BVContainer

I did not see any links from Xiaomi sensors to the Ikea plugs
What type of Xiaomi sensors shows the lines from them to the plugs (routers)?

pergolafabio, 2019-11-06:
i only have the xiaomi aqara pir sensors at home, no others
i have them over a year now, and since now i think 2 months i have a stable system since i have the ikeas and the paired xiaomi ... verry satisfied

pergolafabio, 2019-11-06:
maybe firmware related? in phoscon i see firmware for ikea smart plug :  2.0.022

closed 2020-03-12 by stale[bot]