dresden-elektronik/deconz-rest-plugin #1644: Adding manufacturer specific Develco commands to cluster 0x0006

PeterLyv, 2019-07-02:
I am new to this, so bear with me... ;-)  I am trying to connect a Develco 1 phase DIN rail smart relay. (switch and power meter) https://www.develcoproducts.com/products/smart-relays/smart-relay-16a-din/ 

Basics work fine although it is recognized as a 'light':
[image]

What I am after is adding a few manufacturer specific commands to cluster 0x0006 (on/off). I guess you see them here listed as 0006 - Unknown:
[image]

 Develco has a great manual  ( https://www.develcoproducts.com/media/3698/smrzb-332-technical-manual-smart-relay-din-16a.pdf) and I tried to create a custom ZCL db xml file from it. Without success however. Nothing changes in the UI. I do not get the Safe mode On or Off functions.

Info from Develco:
[image]
(more on page 10 of the manual)

My best effort (develco.xml)

```
<?xml version="1.0" encoding="UTF-8"?>
<zcl>
	<domain name="Manufacturer Specific" useZcl="true"
		description="Manufacturer specific clusters.">

		<!-- Develco -->
		<cluster id="0x0006" name="Develco - Specific clusters" mfcode="0x1015">
			<description>Develco Specific clusters.</description>
			<server>
				<attribute id="0x0000" name="Safe mode Off" type="u8" default="0" access="rw" required="m" showas="hex">
					<description>Disable = 0, 1-255 minutes</description>
				</attribute>
				<attribute id="0x0001" type="u8" name="Safe mode On" required="m" access="rw" default="0">
					<description>Disable = 0, 1-255 minutes</description>
				</attribute>
			</server>
			<client>
			</client>
		</cluster>
	</domain>
</zcl>
```
I am suspecting I am doing something wrong in re-defining attribute 0 and 1 on cluster 6. They already exist, but I do not know how to handle this any other way. Googled for a few hours, but I am totally stuck. Any help appreciated!!

ebaauw, 2019-07-02:
You should add additional manufacturer-specific server commands to the existing cluster definition.  Be sure to specify the parameter.  Don’t change the attributes nor the cluster definition.  Don’t define a new cluster.

I’m not sure deCONZ will like standard and manufacturer-specific commands with the same id - I’ve never tried this, but I do know it doesn’t like standard and manufacturer-specific attributes with the same id.

PeterLyv, 2019-07-02:
thanks @ebaauw for taking the time to help me understand (also a Dutchie! we probably do not live more than 15km apart)

I removed my xml, and added this code to the on-off cluster definition in general.xml:
```
		<command id="0x0000" dir="recv" name="Safe mode Off" required="o">
				<description>Set Off mode after x minutes on Develco switches. The timer can be reset by sending an on command before the timeout occurs. When you keep sending an on command within the timeout period the device will never go to the safe mode. Will also cause the unit to go to Off state when powered up.</description>
				<payload>
					<attribute id="0x0000" type="u8" name="Delay (min)" required="m" default="0x00">  
						<description>Disable = 0, 1-254 = minutes delay</description>
					</attribute>
				</payload>
			</command>
			<command id="0x0001" dir="recv" name="Safe mode On" required="o">
				<description>Set on mode after x minutes on Develco switches. The timer can be reset by sending an off command before the timeout occurs. When you keep sending an off command within the timeout period the device will never go to the safe mode. Will also cause the unit to go to to On state when powered up.</description>
				<payload>
					<attribute id="0x0000" type="u8" name="Delay (min)" required="m" default="0x00">
						<description>Disable = 0, 1-254 = minutes delay</description>
					</attribute>
				</payload>
			</command>
```
It works sort of, I see the command showing up in Deconz in the on-off cluster info, I can fill in the delay value. When I send it I receive a success status, but it ignores the delay value I try to send along, and acts like i have send the regular command 0x0000 or 0x0001. So it just triggers the normal on-off commands not the manufacturer specific functions I am after. eventhough I (try to) send the uint value along.

[image]

Any obvious things I am doing wrong here, or am I entering uncharted territory here with these custom commands with the same id's? (which I find really strange design to begin with to be honest)

ebaauw, 2019-07-02:
You need to add the manufacturer code.

PeterLyv, 2019-07-03:
The documentation never mentions manufacturer code, so I had to guess how this shall be handled. I ended up trying the mfcode attribute on all xml elements, but but it still acts like an on/off switch so no cigar...

```
						<command id="0x0000" dir="recv" name="Safe mode Off" required="o" mfcode="0x1015">
					<description>MFC Set Off mode after x minutes on Develco switches. The timer can be reset by sending an on command before the timeout occurs. When you keep sending an on command within the timeout period the device will never go to the safe mode. Will also cause the unit to go to Off state when powered up.</description>
					<payload mfcode="0x1015">
						<attribute id="0x0000" type="u8" name="Delay (min)" required="m" default="0x00" mfcode="0x1015">  
							<description>Disable = 0, 1-254 = minutes delay</description>
						</attribute>
					</payload>
				</command>
				<command id="0x0001" dir="recv" name="Safe mode On" required="o" mfcode="0x1015">
					<description>MFC Set on mode after x minutes on Develco switches. The timer can be reset by sending an off command before the timeout occurs. When you keep sending an off command within the timeout period the device will never go to the safe mode. Will also cause the unit to go to to On state when powered up.</description>
					<payload mfcode="0x1015">
						<attribute id="0x0000" type="u8" name="Delay (min)" required="m" default="0x00" mfcode="0x1015">
							<description>Disable = 0, 1-254 = minutes delay</description>
						</attribute>
					</payload>
				</command>
```

SwoopX, 2019-07-03:
For me personally, I'm surprised to see that that the manufacturer code has been set correctly under node info. For the smoke sensor, I always get 0x1135 (which should be DDEL if I'm not mistaken), no matter what I do.

ebaauw, 2019-07-03:
What mac address does the _Node Info_ panel display?  Most likely, the GUI thinks you selected the RaspBee or ConBee.

ebaauw, 2019-07-03:
> It almost seems the mfcode is not taken into account.

I scanned general.xml, but couldn’t find an example of a manufacturer-specific command.  Maybe deCONZ GUI doesn’t support these, @manup?

Alternatively, You can try setting these through https://github.com/ma-ca/deconz-cli-plugin.

PeterLyv, 2019-07-04:
I came to the same conclusion regarding mfcode on command elements. Hope @manup can shed a light on that. The node info panel shows:
[image]
Not sure what the MAC address is. The IEEE field contents?

regarding the CLI plugin. Thanks for the pointer, I had a quick look, but I just do not have the time to dig into that at the moment. It seems to add a whole new layer of complexity. I bought the Conbee because I hoped it would give me relatively easy access to this sort of data with deCONZ.

SwoopX, 2019-07-04:
> What mac address does the Node Info panel display?

@ebaauw It definitively shows the Develco MAC in my case. Got some new insights but will open a separate issue for that.

> Not sure what the MAC address is. The IEEE field contents?

@PeterLyv Jep, it is. Are you able, by any chance, to sniff the traffic? How did the join procedure go, smoothly (e.g. without any deconz restarts)?

PeterLyv, 2019-07-04:
>  Are you able, by any chance, to sniff the traffic? How did the join procedure go, smoothly (e.g. without any deconz restarts)?

I am a novice in this field. I am running deCONZ in the ready-made raspberry pi image. I only have one Conbee, so I guess sniffing is not going to work then right? That's what I got from reading about Zshark

SwoopX, 2019-07-04:
Too bad, was worth a try :) I'd have been interested in the join process and the corresponsing node descriptor response in that sense. 

You can sniff with the conbee, but it can't be gateway as well at the same time

closed 2019-11-08 by stale[bot]