dresden-elektronik/deconz-rest-plugin #1645: Commercial Electric downlights randomly blinking

hgelpke, 2019-07-02:
I have 4 of the Commercial Electric retrofit downlight (model 53170161) in my kitchen and when I turn them on at the wall switch at least one will blink for a quick second. After an extended period of time it will happen again but I'm not sure if it's the same fixture or not since it happens rapidly. 

I don't recall having this issue when I was experimenting with Zigbee2Mqtt so I was hoping there might be a Deconz solution.

How can I even try troubleshooting this? Is there any info I can get via VNC?

Smanar, 2019-07-03:
Hello, when you say "turn them on", it mean "send zigbee on command" or "power on" ?
Are you suing pure phoscon or an other home automation application ?

hgelpke, 2019-07-03:
Right now they are on a wall switch so when i power them on at the wall,
there is usually a pause and then a blink. If left on long enough,
sometimes one will blink again.

Smanar, 2019-07-03:
So you can try with deconz off (turning off the gateway) to see if it's an hardware problem or a network (zigbee) problem.

hgelpke, 2019-07-03:
When I had them paired with Zigbee2Mqtt I didn't see the issue which is why
I bring it up.

Smanar, 2019-07-03:
Yep, I don't say deconz can't do nothing, but for the moment IDK if the problem is from deconz or just not corrected by deconz (and was by zigbee2mqtt)

hgelpke, 2019-07-03:
I'm using Deconz/Phoscon through Hass.io but also have access via VNC so I can pull data if needed. 

What I think might be happening is that these bulbs don't have repeater capability so I've put a few repeaters around the house and I'm wonder if they are losing their connection and subsequently re-connecting which is causing the brief flash/blink.

Smanar, 2019-07-04:
If you power off your bulb, for sure they loose their connection, (but not leave the network).
But I see so much things to check:
- Theses device have "power on" configuration ?
- HASS have a setting for that ? On my side for exemple as ikea bulb haven't this feature, I have made one DIY.
- Have you tried with your 4 device in the same group, then not a group.
- Tried with hass disabled, tried with deconz disabled.

On my side, my bulb have "on" state after power on, so even with deconz disabled, if I power on my bulb, even without zigbee network, they turn on, and only that (and after rejoin the network, but only that is not enought to make it flashing). To have a flashing one, they need a zigbee command somewhere.

closed 2019-11-08 by stale[bot]