dresden-elektronik/deconz-rest-plugin #1647: In some cases no sliders (for brightness) or can not switch to color temperature slider.

vandalon, 2019-07-03:
Hi,

My Ikea bulps don't show the brightness slider in phosocon.
Also, in groups with a plug (in this case a OSRAM Lightyfi flug) I can not change the sliders to color temperature, when pressing the "Color Temperature" button, nothing happens. 
This happens within the group, and also the scene editor.

Cheers, and keep up to great work!

vandalon, 2019-07-03:
Follow up, after removing the osram plug I still can't switch to color temperature settings. Other groups work fine, so i'm  not quite sure why this is.

vandalon, 2019-07-03:
I also just noticed the brightness is no longer pushed via websocket with this ikea color bulbs:

```
< {"e":"changed","id":"1","r":"lights","state":{"alert":null,"on":true,"reachable":true},"t":"event","uniqueid":"00:17:88:01:02:6d:1b:e2-0b"}
< {"e":"changed","id":"45","r":"lights","state":{"alert":null,"bri":126,"colormode":"xy","hue":0,"on":true,"reachable":true,"sat":0,"xy":[0.4734,0.3961]},"t":"event","uniqueid":"d0:cf:5e:ff:fe:2a:b6:06-01"}
```

1st bulb: Philips Hue LWB010
2nd bulb: TRÅDFRI bulb E27 CWS opal 600lm

vandalon, 2019-07-03:
Bulbs that seem effected with not having the brightness slider:

- Philips Hue LWB010
- Philips Hue LWB006

The color temp issue I only find with the one TRÅDFRI bulb E27 CWS opal 600lm I have. 

Other bulbs I use are philips hue color bulbs and innr bulbs color and white. The al work perfectly

lettore, 2019-07-03:
I confirm that I'm also missing Brightness sliders for LWB010.
Raspbee version 26330500 Phoscon 2.05.66

ebaauw, 2019-07-03:
> The color temp issue I only find with the one TRÅDFRI bulb E27 CWS opal 600lm I have.

That light doesn't support colour temperature.

> Bulbs that seem effected with not having the brightness slider:
> - Philips Hue LWB010
> - Philips Hue LWB006

That's a bug in 2.05.66: see #1616.

closed 2019-11-07 by stale[bot]