dresden-elektronik/deconz-rest-plugin #1649: Error: (conbee) Bad API KEY domoticz

arifiqbalpk, 2019-07-03:
can someone help me to fix this issue, i am newbie in home automation. i am facing API key issue while configuring Deconz gateway with domoticz home automation.

019-07-03 13:01:56.361 Error: (conbee) Bad API KEY !
2019-07-03 13:02:06.382 (conbee) onHeartbeat called
2019-07-03 13:02:06.382 (conbee) ### Initialisation > config
2019-07-03 13:02:06.382 (conbee) ### Request config
2019-07-03 13:02:06.382 (conbee) Send Command /api/1234567890/config/ with None (0 in buffer)
2019-07-03 13:02:06.382 (conbee) Making Request : http://127.0.0.1:80/api/1234567890/config/ with params None
2019-07-03 13:02:06.402 (conbee) Classic Data : {'apiversion': '1.16.0', 'modelid': 'deCONZ', 'name': 'conbee', 'factorynew': False, 'swversion': '2.5.64', 'bridgeid': '00212EAAAF0325F1', 'replacesbridgeid': None, 'mac': 'b8:27:eb:5f:52:15', 'datastoreversion': '60', 'starterkitid': ''}
2019-07-03 13:02:06.402 Error: (conbee) Bad API KEY !

Smanar, 2019-07-03:
Hello, you are on wrong Github ^^, there is a github just for this plugin.
To have API key you have 2 way
1 - https://dresden-elektronik.github.io/deconz-rest-doc/configuration/#aquireapikey
2 - Use the API_KEY.py utility, Go on plugin folder and use "python3 API_KEY.py", you will have exemple commands

If you need more explanation about this plugin, pls use this github https://github.com/Smanar/Domoticz-deCONZ

Smanar, 2019-07-05:
Have you solved your problem ?

arifiqbalpk, 2019-07-05:
Thanks smanar- Let me check and update

closed 2019-11-09 by stale[bot]