dresden-elektronik/deconz-rest-plugin #1650: Problem for paiting with raspbee.

Smanar, 2019-07-03:
Hello, I have a user with some problem with his hardware. He have
- 2 raspbee (new one) with pre-installed image (with GUI)
- 4 Xiaomi door sensor.
And impossible to pair them, he is using the good procedure, he have the 3 blink + 2 blink on the device, but nothing appear in Poscon on Deconz.

He have tried too the sensor with wireshark, and his config seem ok (channel 11)
We haven't tried log yep, but as deconz have no reactionat all, not sure it will be something on log ?

Some screenshots
[image]
[image]

Someone have an idea ?

bdickhaus, 2019-07-03:
Is he "opening the network" from Phoscon? Sensors->Add New Sensor->Other

The sensors need to be within 10 meters, more or less of the Raspbee (or a router).

Smanar, 2019-07-03:
Yep, and for distance, sensors are not in definitive place, he is just making tests ATM, so he is making tries close to it yes.

Smanar, 2019-07-04:
He is the only one who already have this kind of problem ^^ ?
If he have only 1 raspbee, I could have said an hardware problem, but he have same problem with 2 raspbee and 4 sensors ...
He will make tries with an ikea bulb, but for now apart from contacting support, I realy don't have idea.

thecem, 2019-07-18:
Did you try to pair the sensor directly after power cycle the Raspi, I have this problem. Only after this a few minutes it is possible to pair the devices. But then with a short push on the pair button.

Smanar, 2019-07-18:
But if it work with a short push, it mean the device is already paired ? not necessarily in phoscon but at least in deconz. He have make tries during 5/6 days, so I think he have tried a power cycle.
BTW, for the moment the user is on holiday ^^, so I m waiting for his return.

thecem, 2019-07-18:
you have to open the network via gui, and then push once short on the device button (if the device is not paired). Best results within 5 min: 
- reset the device: (push button for 5 sec, wait for the last flicker)
- Power cycle the ConBee / RaspBee 
- Open Network in Gui
- push the pair button on device

tested with all Xiaomi Sensors/ Switches.

Smanar, 2019-07-18:
I will repeat that, I think he will be back next week.
But I have a doubt with your method, I m sure just before the power cycle, you can see something in deconz, perhaps it's not working, but you probably have something new, and in his case, nothing change in deconz.
Anyway in all case, I will keep you in touch, thx  :)

Smanar, 2019-08-09:
Ok so he have find himself the problem.

> Long story short, 2 different raspbee setups, my own headless version and the image file from Dresden’s site both with a raspbee installed, neither worked, loads of things tried to sort it but no luck then today total success.
> I have always been very cautious about using an SD card for long term raspberry use so always use a mini pci-e (msata) ssd on a converter, the makerhawk X850 to be exact, this happens to be on both my installs and it appears it generates loads of interference right where you don’t want it !!
> 
> If I moved the SSD adapter via a .5 metre usb cable or extended the raspbee header via a cable so there was a few inches gap everything sprang to life.
> 
> So if anyway is having the same issues you now know how to fix it.

closed 2019-08-09 by Smanar