dresden-elektronik/deconz-rest-plugin #1653: Reasonable node descriptor response?

SwoopX, 2019-07-04:
Hi,

I'm still fighting a bit with the Develco smoke sensor integration. I was all the time wondering why the node info panel always showed the wrong manufacturer code 0x1135, which should be 0x1015 instead (#1534).

I sniffed the traffic and learned, that all node descriptor requests were answered with attributes being all 0's, except for the device type. I'll upload a screenshot shortly. Is that reasonable?
To me, it looks like the current values which have been set are some deconz default values. That might also explain my difficulties during join and other observations.

Any thoughts on this one?

Also, I'm beginning to believe that the device could have some firmware issues. Based on available documentation, the latest I found describes clusters which are currently not present. Maybe I can get vendor support to get the latest one.

Cheers!

ebaauw, 2019-07-04:
The endpoints and clusters are advertised in the simple descriptors.  You can force-read these in the deCONZ GUI using the left of the two dropdown circles on the right side of the node.  Would be interesting to sniff the response and see if that’s funny as well.

The device is by (a.o.) manufacturer code, so that would definitely explain pairing issues.

Before concluding that this is a faulty device, I’d double-check that it has joined the network and uses the right encryption key.  Does the sniffer decrypt the responses allright?

SwoopX, 2019-07-04:
Well, simple descriptors look ok to me, but they're only fully collected if I manually do a force-read. As mentioned before, node descriptor requests look fishy...

[image]

closed 2019-11-08 by stale[bot]