dresden-elektronik/deconz-rest-plugin #1655: Change API access URL

rugarci, 2019-07-04:
HI,

I'm running deconz 2.05.66 with docker on a rpi2

I want to access to it via Nginx with:

` location /pwa/ {` 
`      proxy_pass http://192.168.8.128:8083;` 
`      proxy_set_header Host $http_host;` 
`      proxy_set_header Upgrade $http_upgrade;` 
`      proxy_set_header Connection "upgrade";` 
`      proxy_set_header X-Real-IP $remote_addr;` 
`      proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;` 
`      proxy_http_version 1.1;` 
`     }`
   
but I have problems with API url, as I can't use /api path because it is already being used by other service

Is there any way of change the API url to another one like for example /deconz/api ?

I would like to add something like:

`    location /deconz/ {`
`       proxy_pass http://192.168.8.128:8083/deconz/api;`
`       proxy_set_header Host $http_host;`
`       proxy_set_header X-Real-IP $remote_addr;`
`       proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;`
`   }`

SwoopX, 2019-07-04:
Well, is nginx working for you with pwa location? I'm also running it as reverse proxy but never managed to have it working correctly with other location paths as /

Couple of users here had the same issue... However, probably more a nginx rewrite question than deconz related.

rugarci, 2019-07-05:
I can access to:
https://mydomain.com/pwa/index.html
that redirects to:
https://mydomain.com/pwa/login.html

And in this page the discover of gateway starts querying

https://mydomain.com/api/config

that fails as the location I have for "/api" points to other service. This is why I want to move the api URL to another one 

If I point the /api also to the deconz service, it works 

`proxy_pass http://192.168.8.128:8083/api;`

closed 2019-11-09 by stale[bot]