dresden-elektronik/deconz-rest-plugin #1657: Some lights (Tradfri) continuously switching between available and not unavailable

JBS5, 2019-07-05:
Some of my Tradfri GU10 lights are continuously switching between available and not unavailable

[image]

deCONZ: 2.05.66
Conbee firmware: 26330500
deCONZ in Docker running on a NUC with Ubuntu
Had this before (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1597), but the issue is here again.

Does anyone recognize this behaviour? 
Any clue why?

JBS5, 2019-07-11:
A few days later this issue has disappeared (again). Not sure why en how.
In the meantime, the deCONZ docker container is restarted but not sure if this is related.

Smanar, 2019-07-11:
But have you problem in working mode ?
Have you sometime a bulb that doesn't light on or off after a command ?

JBS5, 2019-07-11:
I do not have problems with switching the light on or off during described behaviour.

Smanar, 2019-07-11:
You are sure it's not a problem from the application you are using for log ?
If the bulb was realy unavailable 50% of the time, you will have problem using it ?

JBS5, 2019-07-12:
> You are sure it's not a problem from the application you are using for log ?

The screenshot in my first post is from Home Assistant where the state of the entitiy is going to unavailable which is coming from the rest API. Also in the Phoscon app, the bulb is marked as unavailable.
> If the bulb was realy unavailable 50% of the time, you will have problem using it ?

Could be, but switching the lights happens up to a few times a day.

AlexMekkering, 2019-07-14:
I'm seeing the same behaviour

JBS5, 2019-07-15:
This issue appears and disappears over the days:

[image]

JBS5, 2019-08-02:
And example of a light where the issue arises and disappears.

[image]

JBS5, 2019-11-30:
This issue still present. Happy to provide any information needed..

closed 2020-04-05 by stale[bot]

JBS5, 2020-04-05:
Still happening..