dresden-elektronik/deconz-rest-plugin #1660: Heiman Gas Sensor HS1CG

cf00, 2019-07-06:
I got a new Heiman gas sensor HS1CG, which is on the list of supported devices, with the intention to add it to my Homeassistant setup.

I managed to add the device to deconz, where it shows up as follows:
[image]

In Homeassistant, I also get it to show up as binary sensor (device_class: motion). 

However, when testing it no alarms are transmitted, i.e. when triggering an alarm the sensor state in Homeassistant never changes and is always "off" (I have no idea, how an alarm would register in deconz and how I could check it?). 

Resetting the sensor and pairing again doesn't help

BTW, I also got a Heiman CO sensor (HS1CA) which works without problems as expected.

I am using deCONZ version 2.05.66 (using the latest marthoc/deconz docker image), ConBee II with firmware 264A0700. Homeassistant version 0.95.4. In this setup I have about 30 lights and plugs, 5 motion sensors, 4 door/window sensors, and the CO sensor mentioned above all working without problems.

I don't know if this is a problem with the Homeassistant component or with deconz, but I would assume it's more likely the latter, so I thought I would ask here first.

Any help or pointers for further debugging would be appreciated.

varnone, 2019-10-28:
Did you manage to solve the problem ?

cf00, 2019-10-30:
Sort of. This did the trick: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1711#issuecomment-515715234

Only remaining problem is, that after every reboot I have to manually read the attributes in deconz again, otherwise Home Assistant shows it as unavailable.

faxioman, 2019-11-19:
@cf00 probably is because I'm a pwa app user, but how can I set "the coordinator mac address to the IAS_CIE_address attribute (0x0010) of the IAS Zone cluster (0x0500)" as reported in the comment?

Thanks!!!

michis0806, 2019-11-28:
Sorry, I don't get it, how do you connect the Heiman to DeConz. How to I set the Heiman to pairing mode? Can I connect it via phoscon pwa?

SwoopX, 2020-02-02:
This post might hold the solution: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2403#issuecomment-581164140

closed 2020-06-09 by stale[bot]