dresden-elektronik/deconz-rest-plugin #1661: Conbee I not appearing when launching deCONZ on a raspberry pi3

Klagio, 2019-07-08:
It was workin gfine, then this morning I installed tuya-covert (another project), which required a reboot.

After the reboot when I alunched deCONZ from the GUI, it starts, but shows only Raspbee, Conbee is not there in the drop down menu.

What do I have to do now?

Klagio, 2019-07-08:
```
pi@pioffice:~ $ lsusb
Bus 001 Device 004: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Klagio, 2019-07-08:
deCONZ 2.05.66

Klagio, 2019-07-08:
Tried to reflash

```

pi@pioffice:~ $ sudo GCFFlasher_internal -d 0 -f deCONZ_Rpi_0x26330500.bin.GCF
rmmod: ERROR: ../libkmod/libkmod.c:514 lookup_builtin_file() could not open builtin file '/lib/modules/4.14.98-v7+/modules.builtin.bin'
rmmod: ERROR: Module ftdi_sio is not currently loaded
rmmod: ERROR: ../libkmod/libkmod.c:514 lookup_builtin_file() could not open builtin file '/lib/modules/4.14.98-v7+/modules.builtin.bin'
rmmod: ERROR: Module usbserial is not currently loaded
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
reset FTDI DM01E4IC (dev: 0)
action: update firmware after 3 ms
flashing 125684 bytes: |=============================|
verify: ....
SUCCESS
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/4.14.98-v7+/modules.dep.bin'
modprobe: FATAL: Module usbserial not found in directory /lib/modules/4.14.98-v7+
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/4.14.98-v7+/modules.dep.bin'
modprobe: FATAL: Module ftdi_sio not found in directory /lib/modules/4.14.98-v7+
pi@pioffice:~ $

```

Klagio, 2019-07-08:
```
pi@pioffice:~ $ GCFFlasher_internal -l
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee
pi@pioffice:~ $ lsusb
Bus 001 Device 005: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@pioffice:~ $

```

Klagio, 2019-07-08:
Why is the COnbee disappeared, what do I have to do? I have no USB HUV, the Conbee is connected to the pi3 directly (I tried to change the USB port to where the Conbee is attachjed, but nothing.

Help!

Smanar, 2019-07-08:
Hello
Can you check in /dev folder if you have a device called "ttyUSB0" ?
Can you check if you are in the "dialout" group with "groups" ?

Klagio, 2019-07-09:
Solved.

I took out the Conbee before rebooting the pi3, and put back in. In this way it worked. Strange

closed 2019-07-09 by Klagio

Smanar, 2019-07-09:
I have see somewhere when you reboot the pi, usb port are still powered, so it don't reboot gateway. So you need to unplug power from pi or unplug/replug the gateway.