dresden-elektronik/deconz-rest-plugin #1662: Bitron Smart Plug reports "unknown" type, breaks Home Assistant deCONZ component

thowi, 2019-07-08:
Hi,

I own a [Bitron Smart Plug](http://bvsite.smabit.eu/index.php/produkte/smart-home-produkte/smart-plug-mit-schaltfunktion/) and in HA it's categorized as a light, while it's actually a switch.

HA has a whitelist for device types to fix this. However, this plug reports type "unknown" from deCONZ.
I filed a [bug](https://github.com/home-assistant/home-assistant/issues/25007) against the HA component, but @Kane610 suggested this to be fixed in deCONZ instead and ping you, @manup.

Note that a similar issue has been reported in #332 ~1.5 years ago. The plug works in general. But apparently the type information is still missing.

Can the type information be added?

thowi, 2019-07-30:
Can I provide any more information for this issue to be resolved?

hpd96, 2019-09-10:
Hello,

still not working using 2.05.69

using Bitron power plug-in unit "902010/28"
http://www.bitronvideo.eu/index.php/produkte/smart-home-produkte/smart-plug-mit-schaltfunktion/

JSON:
```javascript
"2": {
    "etag": "ef34257694f525580286b96f6ad76a50",
    "hascolor": false,
    "manufacturername": "Bitron Video",
    "modelid": "902010/28",
    "name": "Lautsprecher Terrasse",
    "state": {
      "alert": "none",
      "on": false,
      "reachable": true
    },
    "swversion": null,
    "type": "Unknown",
    "uniqueid": "00:12:4b:00:0e:f5:bf:9e-01"
  },
  ```
#332 

please check. thanks

hpd96, 2019-09-10:
#332

closed 2020-01-15 by stale[bot]