dresden-elektronik/deconz-rest-plugin #1663: Conbee II not recognized

itis80, 2019-07-09:
I have a new Conbee II that is not recognized in Hass.io (Raspberry Pi 3b+), another Raspberry 3b+ or Windows 8/10. With Windows 8/10 nothing happens and Gcfflasher doesn't pick up any device and nothing is showing up in device manager. I have tried with extension cable as well but no difference.

It shows up in Raspbian as Raspbee unit at port /dev/ttyAMA0 and when I try to flash the firmware I get the following error message:

using firmware file: deCONZ_ConBeeII_0x26490700.bin.GCF
ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn:
dev /dev/ttyAMA0
Reboot device /dev/ttyAMA0 (RaspBee)
serial port open device
uart reset failed, try RaspBee reset
action: reset device RaspBee
close serial port
GCF_ResetDeviceDone
action: update firmware after 2 ms
serial port open device
could not sync with bootloader

Is the software on the device somehow corrupted or is there anything I can still try? I tried to contact the support but no answer so far..

manup, 2019-07-09:
Can you please post the output of `lsusb` on a Linux machine.

itis80, 2019-07-09:
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I believe nothing there - same output as without the Conbee II

manup, 2019-07-09:
Hmm indeed it's not listed there, it might have a loose soldering point that usually connects the usb-stick with the printed circuit board. It's pretty rare, but we had two cases where this was the problem and needed to re-solder it.

I'd suggest if the ConBee II isn't listed on any system, to get a refund and replacement.

itis80, 2019-07-09:
Allright sounds like there is a problem with the stick. I try to get it returned via Amazon.de and order a new one!

closed 2019-07-09 by itis80