dresden-elektronik/deconz-rest-plugin #1664: Wall Switch Terncy (xiaoyan) support

cacherocks, 2019-07-09:
I have a zigbee terncy (xiaoyan) brand wall switch. One button for the load, 3 buttons for the automation of events, without a zero line. In the deconz is not added at all, either through the phoscon or through the touchlink.
Are there any other options how else can it be added to support deconz?
A similar wall switch, but with a zero line from a friend perfectly added in the form of a lamp.
Manufacturer support writes that it works at all zigbee frequencies.

[The official page on the manufacturer's website](https://xiaoyan.io/en-us/wallswitch-withneutral/)

cacherocks, 2019-11-08:
Upd. The request to add a switch is relevant.

Mimiix, 2020-06-05:
@manup is this solved?

closed 2020-07-03 by stale[bot]