dresden-elektronik/deconz-rest-plugin #1666: Question: Orvibo ZigBee CR11S8UZ and Deconz

SAOPP, 2019-07-10:
Hi guys!

I'm sorry for opened this topic, but I can't found any info about compatibility of this switch with Deconz.

I plan to order it right now - but I would like to get at least some information regarding working with Deconz, can someone share any infos? Thanks in advance guys!

This is the official page of the switch: https://www.orvibo.com/en/product/tie.html

SAOPP, 2019-07-18:
So guys any no info regarding this switch?

SAOPP, 2019-07-18:
@Hypfer thanks homie, but I don't have it, and I just try to get some infos because I want to buy it. Of course if I get this switch I share all debug infos about it if this stuff doesn't work with deconz right now.

SAOPP, 2019-07-19:
Guys, I gotted some information about this Tie switch connected to the Deconz.

Here we go:

[image]

[image]

Also, what else we need to be future implementing this connection? Best regards to all Devs guys!

Smanar, 2019-07-19:
They need cluster list and Basic Cluster in detail.

SAOPP, 2019-07-19:
How I can get this list and this details, please?

SAOPP, 2019-07-19:
[image]

SAOPP, 2019-07-19:
[image]

SAOPP, 2019-07-19:
Here is on?

SAOPP, 2019-07-19:
I post some pics of it, for understanding what about we are talking :)

[image]
[image]
[image]
[image]
[image]

Smanar, 2019-07-19:
Yes, it's that, but I don't understand how work your device, I don't reconize your clusters. The only one that can be used is "unknow" ...

SAOPP, 2019-07-19:
Could u please explain, what need to do else for better information of it? Thanks.

Smanar, 2019-07-19:
Nothing you can do I think.
If your device realy have only the cluster 0017 to "communicate", someone need to search how works its attributes.

SAOPP, 2019-07-20:
@Smanar Ok thanks, but is that enough for someone to add this device? I mean, is the current data sufficient for a device to be expected in the future? :)

Big thanks to all guys.

Smanar, 2019-07-20:
I would not want to be a bad omen ^^, But for me, you need a motivated user to use a zigbee sniffer to "decode" how works this cluster, on this project or another zigbee project.
So for me it's not for tommorow, but I m not a specialist.

SAOPP, 2019-07-20:
Okay, thank u for replys homie. I think we are get startred the Tie way to the Deconz. ;)

closed 2019-11-24 by stale[bot]

kelvin5290, 2019-12-11:
Hi, May I know is this repo helpful?
https://github.com/Koenkk/zigbee-herdsman/issues/84

SAOPP, 2019-12-11:
Btw yep, I also saw this topic about konkee...

kelvin5290, 2019-12-12:
I own CR11S8UZ too and I am thinking is it useful for deconz to support this device.