dresden-elektronik/deconz-rest-plugin #1667: New Device Request: Develco magnetic switch w/temperature sensor

pivot69, 2019-07-10:
I am trying to get this Develco magnetic switch to work and I have managed to get it included in the network.

Here's is all the info I could get:

[image]
[image]
[image]
[image]
[image]
[image]

Most of the clusters are either invalid or with general info like this:
[image]
[image]

What I really need is to have a way of reading change status of the switch, when it happens.
It would also be nice with temperature reading, and of course battery status (it seems to be wrong now, they are fresh batteries).

If there is any more info I can send you to add this device just let me know :-)
Thanks!

SwoopX, 2019-07-10:
Yay, another guinea pig :)

Could you please restart deconz and check if the node info panel changes (#1644 )?

General Develco support has already merged into the code. If you feel funny, you may compile the latest version and give it a try or you wait for the next official release.

pivot69, 2019-07-10:
I’m happy to be a lab rat, especially when I can collaborate with the scientist to get the results needed 😎

Restart the device (from deconz GUI), the service or the raspberry pi itself?

SwoopX, 2019-07-10:
:)
Deconz service should suffice...

pivot69, 2019-07-10:
Now there is more! :-)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2019-07-11:
Hm, first screenshot is not exactly what I expected. Funny thing is, the device has an IKEA manufacturer code but that is without the changes made for Develco. My smoke sensor, however, is mentioned as Dresden Elektronik device, even with the changes. The corresponding zigbee traffic doesn't show anything reasonable as mentioned in the referenced issue of my first reply. I'd love to have an opinion from @manup on this one though.

As I said, you may want to compile the current version of the plugin following the instructions in the readme. There's a chance getting it to work then (delete and new join). Otherwise, a bit patience is required.

pivot69, 2019-07-12:
Strange that it identified it as a Router and not an End device too, right? 
But now its changed:

After triggering it (it seems unresponsive to the actual magnet switch, so i had to wake it by holding the sync-button until it started blinking) I could read some more via the clusters. It is alive for about 30 seconds when i trigger it like this.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

I cant seem to find the attribute for the actual magnet switch anywhere though... Is there something thats not being detected? I notice the c0c9 level control switch section being pretty much unknown...

I could compile the new version of the plugin, but first I think its best to cover all this so I dont break stuff for no purpose :-)

SwoopX, 2019-07-13:
> Funny thing is, the device has an IKEA manufacturer code but that is without the changes made for Develco. My smoke sensor, however, is mentioned as Dresden Elektronik device, even with the changes.

What I mentioned earlier seems to be some "ghosting effect". I'd assume you also have some IKEA devices running...? No big deal, I hope

> I could compile the new version of the plugin, but first I think its best to cover all this so I dont break stuff for no purpose :-)

True story. I can fiddle with the code to support it, but I'd prefer to do that in my own fork first before pulling it. Also, it will only be exposed via REST API, meaning that you will most likely don't see it in Phoscon...

pivot69, 2019-07-14:
Thanks for your follow up on this 🙂

Yep, got about 10 ikea remotes and maybe 35-40 ikea lights (float panels, spots and bulbs).

But what’s the next logical step now? Anything more I can share or try? 

Don’t want to way for a future fix, and would much rather try something and contribute with the development of this now. I have been needing these magnetic switches for a while now (got about 15-20 of them, ready to be used).

Also gonna get my Danalocks working, some smart outlets - and potentially some flammex smoke detectors... but that’s another story and another thread 😎

SwoopX, 2019-07-14:
Well, point is, it needs to be tested.

**Theoretically**: With the change I made in my fork, the device should generally be supported. As to my understanding, the switching events should come in via IAS Zone cluster. Based on your latest screenshots, you've read the IAS Zone cluster attributes, but it reported not enrolled. I would expect this to change due to my commit. As a result, a sensor node in the REST API should be created and the device should also do/show switching.

Now the BUT: As I learned from my smoke detector, the device is a little bit subborn. First of all, the join is a pain in the ... and I have to restart deconz twice to have all nodes created and identified properly. Secondly, I get a rubbish response while reading the node descriptor, as all values are zero when I sniff the traffic. I'm not a zigbee expert, but this cannot be normal ( #1653). I'd also assume that this is the root cause for the join difficulties.
Now, I know that my device has an older firmware which can be concluded based on available documentation (technical manuals from develco and other sites). The latest (from develco) shows clusters which I do not have. My theory is that the firmware update could eventually fix the node descriptor as well as some other issues I experienced (e.g. temperature not being reported at all, not too sure if battery reporting is ok). Already reached out to Develco support but no response. Turned to Cozify support as well, as their gateway supports the sensors, but they seems to be not allowed to share the firmware.

I couldn't find any technical manual for your device on a quick search, however, I suspect it has the same issue with the node descriptor as well. To be honest, I currently do not see a high chance to get your sensors to work. Sniffing the zigbee traffic could give some further insights. Bought a CC2531 and another Raspbee for testing purposes and being able to play around if required.

As of now, I'm also a bit hesitant to buy another smoke sensor. Risk is too high that it has the same firmware for 50€ or the experienced behaviour persists. Too expensive for a try. But in #1424, I've seem that Develco has also devices which identify correctly ;)

pivot69, 2019-07-15:
> Well, point is, it needs to be tested.
>
> **Theoretically**: With the change I made in my fork, the device should generally be supported. As to my understanding, the switching events should come in [...]

Yeah, I was thinking the same and I have tried sending it commands to get it enrolled. Only reply I have managed to get is something like "UNSUP_CLUST_CMD" (if I remember correctly, I'm not near the device now so I cannot check...) even when trying to use the "standard" command of "Initiate Normal Operation Mode"

[image]

> **Now the BUT**: As I learned from my smoke detector, the device is a little bit subborn. First of all, the join is a pain in the ... and I have to restart deconz twice to have all nodes created and [...]

The switch I have is actually a rebranded Develco product, sold by a company called "Home Control" in Norway with a product line called "Wattle". (The ones I have is the previous models, not the one your find on their site)

They use a lot of the Develco products with their controller and they are both eager to learn and has been very helpful with my support requests. It might be worth getting in touch with them to ask your in depth tech questions, and they might have valuable info to share about the Develco products and getting them working properly. They have been at it a while. 

https://wattle.com/support-main/contact-us/

> I couldn't find any technical manual for your device on a quick search, however, I suspect it has the same issue with the node descriptor as well. To be honest, I currently do not see a high chance [...]

If you want I can send you one of my (new, still in the box) switches. I'll pay for the postage too. They are small and light, and should be quick and cheap to send. If you get it working you have a new magnetic switch for your network too ;-)

Can you DM me your shipping address? 
(is there even a direct message function on github??)

ebaauw, 2019-07-15:
> I have tried sending it commands to get it enrolled. Only reply I have managed to get is something like "UNSUP_CLUST_CMD" 

I've yet to see a device that has implemented this command.  To enroll an _IAS Zone_ device manually, write the mac address of the RaspBee or ConBee to the _IAS_CIE_Address_ attribute.  The REST API plugin should take care of this automatically, once the device has been whitelisted.

SwoopX, 2019-07-15:
> They use a lot of the Develco products with their controller and they are both eager to learn and has been very helpful with my support requests. It might be worth getting in touch with them to ask [...]

Yeah, great. I wanted to walk away from Develco and you keep me tempted... 😋  Are they also still fine if you don't use their gateway? I suppose you own one of those [https://wattle.com/product/magnetic-sensor/](https://wattle.com/product/magnetic-sensor/). Also seen that they have the somek sensors as well. I'd love to get my hands on any technical documentation as well as the firmware files...

SwoopX, 2019-07-16:
Some good news.

Out of a sudden, attribute reporting started to work (temperature and battery). No idea why, but good that is does. Let me see what additions are required for the switch and prepare a PR. However, you still need to test it.

pivot69, 2019-07-17:
> Some good news.
> 
> Out of a sudden, attribute reporting started to work (temperature and battery). No idea why, but good that is does. Let me see what additions are required for the switch and prepare a PR. However, you still need to test it.

NICE! :-D

I see that you have requested a merge. How long will it take before its merged? Tried running an upgrade now, but as I can see its still open so no changes on the source yet.

pivot69, 2019-07-17:
> I've yet to see a device that has implemented this command. To enroll an _IAS Zone_ device manually, write the mac address of the RaspBee or ConBee to the _IAS_CIE_Address_ attribute. The REST API [...]

That doesnt seem possible. Its greyed out, not accessible for writing (see new screenshot, and earlier screenshot). Even when I try writing other places where the attribute has a rw settings I cannot write anything and it just gives me a "writing failed"...

[image]

Or am I missing something? How can I write the mac-address to the attribute?

pivot69, 2019-07-17:
> Yeah, great. I wanted to walk away from Develco and you keep me tempted... 😋 Are they also still fine if you don't use their gateway? I suppose you own one of those [...]

They are fine with it, but regular first line support cant offer me much support when using other gateways than theirs. But their technicians should be able to, and with your knowledge you might also be able to contribute something back so they also gain something. 

You can contact them through their regular support. I have a reference/name you can have so that you get in touch with the guy I've had contact with. He can probably direct you or help you directly.

But I guess I should rather send it through some kind of DM and not share it open here ;-)

I have the same magnetic switch, but its their previous version. Afaik it's the same on the inside.

SwoopX, 2019-07-17:
> Or am I missing something? How can I write the mac-address to the attribute?

Jep, the whitelisting. That's part of my commit.

> I see that you have requested a merge. How long will it take before its merged?

Good questions. Just picked up manup is on vaccation...
I'd say let's wait for the merge and ideally an official release. We could use the discord channel mentioned in one of the latest issues.

closed 2019-11-21 by stale[bot]