dresden-elektronik/deconz-rest-plugin #1668: Power-on default osram lights

intruder7777, 2019-07-10:
as mentioned in issue #102 is it possible to implement this functionality for osram lights?

quotschmacher, 2019-07-26:
i would think this would be very useful. maybe even automated - whenever a parameter changes the new values could be stored as default to the bulbs (only the changed values). but this would be the next step - the first would be a simple button to store values (maybe even a feature for the deconz rest api)

ebaauw, 2019-07-26:
Not sure that's a good idea.  The power-on settings are stored in non-volatile memory, which typically can be written only a limited number of times.  Writing it every time you change the light state might severely shorten the life time of the light.

quotschmacher, 2019-07-26:
depends on how many write cycles the memory supports during life time and how often i want to write... but you are right. 
but as long as one can't wirte vie deconz that's not the question.

lrybak, 2019-07-26:
Writing the power-on defaults each time the switch/light is toggled is not good idea as it can wearout the momory. Fully agree here.
But there should be an option to save power-on defaults on demand for paticular node. At the momment my OSRAM Flex strip is not usable as each time once power is lost (unfortunatelly common thing in my area) it turn on with furious bright light.

intruder7777, 2019-07-26:
I think a possibility to store the actual  values as power-on default is enough . it is possible to do it with the osram gateway and the lightify app so it should be possible to do it with deconz

lrybak, 2019-07-26:
> I think a possibility to store the actual values as power-on default is enough . it is possible to do it with the osram gateway and the lightify app so it should be possible to do it with deconz

Can you point me how to do that with deconz?

intruder7777, 2019-07-26:
no sorry, actually it is not possible with deconz. it should be implemented by developer

closed 2019-11-30 by stale[bot]