dresden-elektronik/deconz-rest-plugin #1669: Request to add NYCE NCZ-3041 Motion Sensor

ScottGA, 2019-07-11:
I saw a couple other version of NYCE motion detectors, but not the 3041.  Please add.  Thank you!!!

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ScottGA, 2019-07-11:
I would also like to add an additional request to include a different type of NYCE sensor in the updated database - the NYCE Door Hinge Sensor Model NCZ-3010.  I pasted the related screenshots below:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ScottGA, 2019-07-23:
Is there anything I can do myself to use the above devices with deconz?

ScottGA, 2019-11-22:
Is there any update on getting this sensor device added?

jensflorian, 2020-01-22:
The NYCE3041 is HA1.2 compliant, so incorporation should be not too difficult. It is now sold in Germany by ubisys: https://www.smarthome-store.de/media/documents/NYCE_Feature%20Sheet_Motion%20Sensor.pdf  Is anyone able to use this in the deconz REST API?

closed 2020-05-30 by stale[bot]

amunamu, 2020-05-31:
I would also like to have support for NCZ-3041