dresden-elektronik/deconz-rest-plugin #1670: Philips HUE lights not discovered

krisjansen, 2019-07-11:
Dear,

I have a set of Philips HUE lights which are connected to HUE brige.
I also have a set of TRADFRI switches.

My intention is to disconnect the HUE Bridge and add all components to my Raspberry PI so the mesh network can be mixed between IKEA & Philips components. 

Alas after resetting my HUE lamps (and even factory reset of the HUE bridge) I can only discover a subset of HUE lamps. My TRADFRI's are correctly in the mesh, only 3 out of 15 HUEs are in the mesh network. The rest I can only discover again via the HUE bridge/app and not via deCONZ/Phoscon.

Do you have any idea how I can force the HUE lamps to be rediscovered by deCONZ/Phoscon App? 

Thanks a lot!
K

Kane610, 2019-07-11:
Have you power cycled the lights after resetting them? That is one possible issue

b2un0, 2019-07-12:
power cycle multiple times during search process helps a lot.

krisjansen, 2019-08-11:
tried everything I could think off...

1. I turned off my Hue bridge
2. Factory reset all my lights (with the switch)
3. Put deconz in discovery
4. power cycled all lamps dozen of times during the search...

Nothing happens.

krisjansen, 2019-08-21:
Finally got it all to reset correctly and join the zigbee cluster.
I learned that resetting the Hue lights via the dimmer doesn't do a proper reset for deconz to discover the lights.
The only reset which works fine is via Touchlink.
Since my bulbs were installed at difficult to reach locations (outside/garden) and couldn't get them close to the RPI, I made the RPI mobile via powerbank and reset them at install location.
Afterwards I had to disover the lights with the RPI still close to the Hue bulb which is then added correctly to my mesh network.

Finally I can complete my home automation :-)

ebaauw, 2019-08-21:
> For the irrigation I am using disasembled Tradfri Outlets, connected to solenoid valves (at 220V AC) which control my irrigation circuits

If you’re using HomeKit, homebridge-hue can actually expose the outlets as valves, see https://github.com/ebaauw/homebridge-hue/issues/410#issuecomment-454503488.

closed 2019-12-26 by stale[bot]