dresden-elektronik/deconz-rest-plugin #1671: Aqara (xiaomi) motion sensor mod support

sorryusernameisalreadytaken, 2019-07-11:
There is a hardware-mod for the Aqara motion sensor to get a shorter check-up time.

I first see this in the smartthings board [here](https://community.smartthings.com/t/making-xiaomi-motion-sensor-a-super-motion-sensor/139806).
Later a user from the 
[homeassistant community create a code-snipped for a custom-component](https://community.home-assistant.io/t/xiaomi-human-body-motion-sensor-timeout/23398/156?u=jamieme).

There should be no problem to get officialy support for this mod, or?
At the moment we need to modificate the seconds in each motion sensor in deconz with a tool like Postman. We need to send a POST request at http://hassioip:40850/api/apikey/sensors/sensornumber /config ; in body (raw) put: {“duration”: 5} - where 5 is refresh time.

I think a not alone with the big wish to enhance the 90 seconds timeout to something lower like the 5 seconds.

00lex, 2019-07-11:
that's awesome!!
isn't it enough to modify the zll.db from deconz to x seconds? I will try it this evening! 

[image]

btw. original duration is 60s. don't know why and when deconz change this to 90s

00lex, 2019-07-11:
until someone make changes to deconz I will run a script on every start of deconz for all sensors. 

`curl -X PUT -d '{"duration":5}' http://127.0.0.1:8009/api/YOURAPIKEY/sensors/33/config`

thank you for the hint.

00lex, 2019-07-11:
Okay this is my final script. 

```
sensors="25 11 29 16 5 33"

for i in $sensors; do
    curl -X PUT -d '{"duration":5}' http://127.0.0.1:8009/api/YOURAPI/sensors/$i/config
done
```

while testing I see one strange thing. one of my aqara sensors has a default duration of 122 seconds

Kane610, 2019-07-11:
You can create an automation in hass that does that with deconz service on every start 👍🏻

00lex, 2019-07-11:
I use pimatic. thanks anyway

sorryusernameisalreadytaken, 2019-10-07:
> 
> 
> Okay this is my final script.
> 
> ```
> sensors="25 11 29 16 5 33"
> 
> for i in $sensors; do
>     curl -X PUT -d '{"duration":5}' http://127.0.0.1:8009/api/YOURAPI/sensors/$i/config
> done
> ```
> 
> while testing I see one strange thing. one of my aqara sensors has a default duration of 122 seconds
> 
> I use a pencil (HB) too, to connect both pins and put some nail polish to on it fix the graphen

I had no duration tag under device. 
modelid  is "lumi.sensor_motion.aq2"

Can I add this duration tag?

I got an error when I try to add your 5 second value.

hendrikmaus, 2019-10-07:
What is your error you get? I have been using the same method successfully with the model ID you mentioned.

sorryusernameisalreadytaken, 2019-10-07:
@hendrikmaus

My error is:
[
    {
        "error": {
            "address": "/sensors/1/config/duration",
            "description": "parameter, duration, not available",
            "type": 6
        }
    }
]

hendrikmaus, 2019-10-14:
I reckon that you added the 'duration' to the URI. Let's review your curl command. It is important to pass the duration using JSON payload to the config  resource. An example was posted above.

ikrowni, 2019-12-16:
Can someone explain exactly what I need to do to get this 5 second timer working with home assistant? I've already soldering the wire onto a few of my motion sensors but I don't know what to do from here.

ikrowni, 2019-12-16:
> You can create an automation in hass that does that with deconz service on every start 👍🏻

Could you share this automation please?

hendrikmaus, 2019-12-17:
To sum it up; after applying the hardware modification:

### Get API Key

Open Phoscon, navigate to the gateway and allow apps to connect:

```sh
# replace '10.0.0.1' with the IP of your deCONZ deployment; port is the default one
$ curl -X POST -s -d '{"devicetype":"curl"}' "http://10.0.0.1:40850/api"
[{"success":{"username":"XXXXXXXXXX"}}]
```
Note down the `username` and remember to replace both IP and username in all subsequent examples.

Now use the `username` as API key to list all your sensors. This is required to get the IDs:

### List sensors

```sh
curl -s http://10.0.0.1:40850/api/XXXXXXXXXX/sensors
```

If you found your sensor ID, this is the request to set the duration manually:

### Set duration for motion sensor

```sh
$ curl -X PUT -d '{"duration":5 }' -s http://10.0.0.1:40850/api/XXXXXXXXXX/sensors/42/config
[
  {
    "success": {
      "/sensors/42/config/duration": 5
    }
  }
]
```

In home assistant, I wrapped the previous command into a `shell_command` and use this `automation` to continuously tell deCONZ to keep this behaviour:

_Note: I run this every 5 minutes, which is probably not required._

```yaml
deconz_set_duration_for_motionsensor: >
  curl -X PUT -d '{"duration":5 }' -s "http://10.0.0.1:40850/api/XXXXXXXXXX/sensors/{{ id }}/config"
```

The `automation`:

```yaml
alias: 'xiaomi_motion_sensor_refresh_duration'
trigger:
- platform: time_pattern
  minutes: "/5"
action:
- service: shell_command.deconz_set_duration_for_motionsensor
  data_template:
    id: '42'
```

Kane610, 2019-12-17:
You should use [Deconz Configure Service in hass](https://www.home-assistant.io/integrations/deconz/#service-deconzconfigure)

ikrowni, 2019-12-17:
> You should use [Deconz Configure Service in hass](https://www.home-assistant.io/integrations/deconz/#service-deconzconfigure)

Could you elaborate?

ikrowni, 2019-12-18:
> ```shell
> curl -X POST -s -d '{"devicetype":"curl"}' "http://10.0.0.1:40850/api"
> ```

im getting this error when trying to run the curl command 

curl -X POST -s -d '{"devicetype":"curl"}' "http://192.168.0.15:40850/api"
through a cmd prompt on windows 

[{"error":{"address":"/","description":"body contains invalid JSON","type":2}}]

ikrowni, 2019-12-18:
thanks i got it working now.

Vissertje1980, 2019-12-20:
I there someone who can tel me how i can make a script in Python to do the PUT requests automatic after reboot? I already installed the requests module on the pi en tried to make a script to set the velue of one sensor to 5 seconds but it did not work.

On my laptop, it works goot with a REST client but i don't no how to doe it automatic on de the pi.

this is the script i treid:

import requests

r = requests.put('http://ip adress:80/api/YOURAPI/sensors/32/config', data = {"duration": 5})

print (r)

i get code 200 back but the value of duration of sensor 32 did not change wen i check it with a GET request on the laptop.

chbndrhnns, 2019-12-26:
> You should use [Deconz Configure Service in hass](https://www.home-assistant.io/integrations/deconz/#service-deconzconfigure)

When I try to send data, I receive the message "parameter duration is not available":

```yaml
entity: binary_sensor.presence_24
data:
  'duration': 5
```

@Kane610 Would you mind sharing your payload?

jyrki69, 2020-01-11:
> > You should use [Deconz Configure Service in hass](https://www.home-assistant.io/integrations/deconz/#service-deconzconfigure)
> 
> When I try to send data, I receive the message "parameter duration is not available":
> 
> ```yaml
> entity: binary_sensor.presence_24
> data:
>   'duration': 5
> ```
> 
> @Kane610 Would you mind sharing your payload?
> 
> _edit_
> 
> I forgot the config key. It works like this:
> 
> ```yaml
> entity: binary_sensor.presence_24
> data:
>   config:
>     'duration': 5
> ```

Are you still getting this to work?
When I try this I get an error, /sensors/config parameter, config, not modifiable.

I read some posts that they had made config read only, but that was from 2017.

Any suggestions?

ebaauw, 2020-01-11:
You need to do a PUT to `/api/`_apikey_`/sensors/`_id_`/config` with a body of `{"duration": 5}`.

See http://dresden-elektronik.github.io/deconz-rest-doc/sensors/.

askanhesse, 2020-03-27:
Hi there.
I'm also running into the issue that I can not setup the duration for the sensor. Is there any way to solve this?

Deconz version 2.05.75 / 8.3.2020

Sample CURL:
`curl -X PUT -d '{"duration":5}' -s "http://127.0.0.1/api/xxxxxxxxxx/sensors/94/config"`

Response:
`[{"error":{"address":"/sensors/94/config/duration","description":"parameter, duration, not available","type":6}}]`

askanhesse, 2020-03-27:
Since I am reading the same behaviour as happened for me in this thread, that people getting the error response:
`[{"error":{"address":"/sensors/{ID}/config/duration","description":"parameter, duration, not available","type":6}}]`

Make sure you used the correct device ID since this sensor is creating two devices in deconz. Choose for that one of type `ZHAPresence`.

sebastianlorenzen, 2020-05-08:
Any news on this? The duration is still reset, after you reboot deconz gateway.

sebastianlorenzen, 2020-06-27:
When will this be implemented?

GitHubStephan, 2020-09-17:
the point is still open. would be fine if you can implement it in deconz

Smanar, 2020-09-17:
It was not corrected by https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3176 ?

GitHubStephan, 2020-09-17:
hello, sorry, i miss this point. only read this thread and see that it is closed without comment. i will test it with my new hardware mod.

oneseventhree, 2020-09-27:
This is still an issue. After the hardware mod it still taking 90 seconds to clear instead of 5 seconds

Mimiix, 2020-09-28:
@oneseventhree @GitHubStephan What version of deCONZ are you on?

oneseventhree, 2020-09-28:
> @oneseventhree @GitHubStephan What version of deCONZ are you on?

This is what I am running

[image]
[image]
[image]

Mimiix, 2020-09-28:
So it should be working @oneseventhree 
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_81_stable

oneseventhree, 2020-09-28:
> So it should be working @oneseventhree
> https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_81_stable

It should but it still takes 90 seconds in home assistant to show as cleared

Mimiix, 2020-09-28:
You still need to configure it yourself using the api.

oneseventhree, 2020-09-28:
> You still need to configure it yourself using the api.

Do you know how I can do this?

Mimiix, 2020-09-28:
Please read up a bit ;)
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1671#issuecomment-573344211

oneseventhree, 2020-09-28:
> Please read up a bit ;)
> [#1671 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1671#issuecomment-573344211)

This is all a bit over my head so I have to take it step by step. How do I find the API key?

oneseventhree, 2020-09-28:
For anyone who struggled with this as much as I did, all you need do AFTER you do the hardware modification is to go into the UI of Home Assistant then go to 'Developer Tools > Services' then type this(but change the entity to yours):
[image]

Code is:
```
entity: binary_sensor.YOUR_ENTITY_NAME_HERE
data:
  config:
    'duration': 5
```

I've restarted HA and turned off my rasberrypi/deconz and it works. Hope that helps someoen because reading up confused me alot :)

Mimiix, 2020-09-28:
As this seems fixed:  I'll close the issue ;)

closed 2020-09-28 by Mimiix