dresden-elektronik/deconz-rest-plugin #1672: Support for Lutron Aurora (Friends of Hue dimmer)

glward, 2019-07-11:
The Lutron Aurora is a new Friends of Hue dimmer (http://www.lutron.com/en-us/products/pages/standalonecontrols/dimmers-switches/smartbulbdimmer/overview.aspx) 

I've paired it with deCONZ but it is not recognized as a switch:

[image]
[image]
[image]

I don't see much in the logs when pairing, but I'm not sure I've turned the correct ones on to show traffic during pairing or after.

glward, 2019-07-11:
I paired it with my Hue bridge and captured the Zigbee traffic as well; I've attached that capture. 
[lutron_aurora_hue_filtered.pcapng.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3384123/lutron_aurora_hue_filtered.pcapng.zip)

glward, 2019-07-14:
I've hacked on the plugin enough to get the dimmer recognized in the deCONZ UI (not in Phoscon), bind it to lights, and (very hackily) generate sensor button events when the center button is pushed (since it's generating brightness events I can send "1002" on lights on and "4002" on lights off, which kind of seems wrong but suits my particular use case :P). I tried to get it to pair like the Hue dimmer does, since it appears to support the same cluster 0xfc00 commands as the dimmer based on my sniffing of it when paired to a Hue bridge, but I didn't get anywhere. I get an "Unauthorized" message when it attempts to bind to cluster 0xfc00 at join, and even after binding manually the dimmer isn't sending any commands at that cluster, just the level cluster. 

I'm happy to provide my code, but it's pretty hacky and littered with my poor understanding of how some of deCONZ works -- I suspect someone more familiar with the code would be better off just starting over.

Smanar, 2019-07-14:
It's holidays period, and there is already lot of device integration issue. So I think a partial unworking code will attract their attention more than a simple issue.

stevmills, 2019-08-01:
@glward I am attempting to do the same thing you've described here and would love to get the code you've had some success with. Have you made any more progress integrating the Aurora?

akonradi, 2019-08-25:
I'm not at all familiar with deCONZ but I'd appreciate having your code if only so I can get my lights working :)

glward, 2019-08-25:
Sure:

[aurora.patch.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3538354/aurora.patch.txt)

It's against the repo from a few weeks ago so you may need to update to an older changeset before applying it.

No promises it will work :P

closed 2019-12-30 by stale[bot]

TFenby, 2020-01-04:
Can this be re-opened? These devices are really useful...

slootjes, 2020-01-06:
I'd like to have this in too.