dresden-elektronik/deconz-rest-plugin #1673: Cant control Philips LWB010

peer69, 2019-07-12:
Starting with 2.05.66 i cant control LWB010 Lamps from Phoscon GUI anymore. The gui only shows the slider for brightness which doesnt seem to work. 
Looking at Homekit using Homebridge-Hue I now see only a switch which still works but i cant dim the lights any longer: 

[image]
[image]

Downgrading so 2.05.65 solves the problem.

ebaauw, 2019-07-12:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1616

peer69, 2019-07-12:
Thanks for the reference. So this should be fixed with .67.

closed 2019-07-12 by peer69