dresden-elektronik/deconz-rest-plugin #1674: Feature request: a scene that turns on/off a light based on a door sensor value change

Klagio, 2019-07-13:
Could this be implemented?

ebaauw, 2019-07-13:
Not sure I understand your use case, but, basically, yes.

Create a CLIPGenericFlag sensor and two rules that trigger on the flag being set.  One rule turns on the light, the other turns it off, depending on the door state.  Then create a scene to set the flag.

Klagio, 2019-07-13:
Hi thanks for answering.
My user case is the following: I have a closet room in which I want that when I enter (door has a Xiaomi door sensor) it turns on the lights (they are connected to a Xiaomi power plug), and off when I close the door (when I am inside I do not close the door). Make sense?

How to create a CLIPGenericFLAG?

ebaauw, 2019-07-13:
In that case, you don’t want to use scenes, and you don’t need the CLIPGenericFlag sensor.  Simply create two rules: one to turn on the plug when the door is opened, one to turn it off when the door is closed.

```json
{
  "actions": [
    {
      "address": "/lights/251/state",
      "body": {
        "on": false
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/256/state/open",
      "operator": "eq",
      "value": "false"
    },
    {
      "address": "/sensors/256/state/open",
      "operator": "dx"
    }
  ],
  "name": "Washroom Door Close"
}
```

You create the rules thru the REST API, by doing a POST to `/rules`, see http://dresden-elektronik.github.io/deconz-rest-doc/rules/.  Similarly you would create a CLIP sensor by doing a POST to `/sensors`.  I use `ph` to interact with the API from the command line.

Klagio, 2019-07-13:
Ok, trying to. Is there any tutorial?
I have difficulties from the start, on how to get the api key

ebaauw, 2019-07-13:
See _Getting Started_ on the website I mentioned earlier.

Klagio, 2019-07-13:
Ok, thanks a lot, I put below the one for open and the one for close. What is operator dx, do I need to put that?

{
  "actions": [
    {
      "address": "/lights/3/state",
      "body": {
        "on": false
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/59/state/open",
      "operator": "eq",
      "value": "false"
    },
    {
      "address": "/sensors/59/state/open",
      "operator": "dx"
    }
  ],
  "name": "Closet Door Close"
}

{
  "actions": [
    {
      "address": "/lights/3/state",
      "body": {
        "on": true
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/59/state/open",
      "operator": "eq",
      "value": "true"
    },
    {
      "address": "/sensors/59/state/open",
      "operator": "dx"
    }
  ],
  "name": "Closet Door Open"
}

ebaauw, 2019-07-13:
> What is operator dx, do I need to put that?

It checks that the value has actually changed.  Otherwise the rule might fire every time the sensor sends a periodic report.  I created my rules a long time (and many versions) ago.  Come to think of it, I’m not really sure it’s still needed, but it surely won’t hurt.

Klagio, 2019-07-14:
Ok, thanks now I managed to add the rules, but had very stange behavior.

Foreword: I am using Home Assistant for all my automations, which is great, but sometimes it hangs, or has problems, so I wanted to double reliability, and put some automation ALSO directly on deCONZ.

SO the RULES are perfect for this.

Problem is applying the rules above did very strange behavior ... and I do not know how to troubleshoot

ebaauw, 2019-07-14:
> Problem is applying the rules above did very strange behavior

Define "very strange".

> how to troubleshoot

- Shutdown Home Assistant to rule out any interference;
- Run `dc_eventlog`, or something similar to see the websocket notifications deCONZ sends out, to monitor the changes to the door sensor and the light.  You might want to filter the output:
  ```
  $ dc_eventlog -s 2>&1 | grep -e /sensors/59 -e /lights/3
  ```
  I actually run `dc_eventlog -s` as a service all the time;
- Monitor `lasttriggered` and/or `timestriggered` of your rules to see when they fire.  Note that the time is in UTC.  Unfortunately they don't emit a websocket notification - still somewhere on the lower part of my todo list.  Again, you might want to filter the output:
  ```
  $ while true; do ph get -al /rules | grep triggered: ; sleep 1 ; done
  ```
- You might change the rule actions from PUTting the `state` of a `/lights` resource to PUTting the `action` of a `/groups` resource, even though the group only contains a single light.  Group commands are sent as broadcasts on the Zigbee network and tend to be more reliable than unicast light commands, especially with (IKEA) lights being marked unreachable when deCONZ loses the route to them.

> The deCONZ RULES are perfect for this.

Yes they are.  I run all my automations involving Zigbee devices on deCONZ rules, using `CLIPGenericFlag` and `CLIPGenericStatus` sensors as variables to keep state.  I've got 284 rules in my production environment of 100 ZigBee devices.   I maintain these rules (and the CLIP sensors) using `ph` and bash scripts, see https://github.com/ebaauw/ph.sh.  I use HomeKit for automations involving non-Zigbee devices (notably my Sonos speakers).  I bridge the two environments by exposing the `CLIPGenericFlag` and `CLIPGenericSensor` sensors to HomeKit.

Klagio, 2019-07-14:
> Yes they are. I run all my automations involving Zigbee devices on deCONZ rules, using `CLIPGenericFlag` and `CLIPGenericStatus` sensors as variables to keep state. I've got 284 rules

PRetty cool, althogh, isn't it hard to maintain without using a better interface? As an example if you just need to change a time (for example turn ON the light not anymore at 8pm but at 8,30pm) its difficult to remember api, ID of light, ID of rule and so on ..

ebaauw, 2019-07-14:
That’s why I use scripts.  I simple delete and re-create all rules, so I have no need for the rule IDs.  I patched the database, re-assigning the IDs of my lights, groups, and sensors to match a number scheme (floor-room-sequence or floor-room-sensortype) that I can remember easily.  `ph` remembers the API key.

Klagio, 2019-07-18:
> Define "very strange".

Its happening with one Xiaomi smart switch, and two door sensors.

The smart switch I found out that goes unavailable in Home Assistant but its there in PHOSCON app, It could be a signal issue, but strange I do have 2 Zigbeee bulb not far from it that are working well (an IKEA wall lamp and a HUE Bulb)

The two door sensor do not work right: especially my front door sensor (for me that sensor is very important for various automations), it stays ON, even when I close the door .... almost all the time: the only way to fix is open/close the dorr 1 or 2 times. In this case signal cannot be an issue, its full of Zigbee devices (60+)

ebaauw, 2019-07-18:
> In this case signal cannot be an issue, its full of Zigbee devices (60+)

Not necessarily.  To what router is the door sensor connected?  Xiaomi sensors don't switch parents.  Make sure to pair them close to where you'll be using them, so they chose a parent router well within range.

Klagio, 2019-07-18:
Ok, the smart switch is connected to another bulb nearby so should be ok (then what is it?)

The front dorr sensor is connected to a parent actually a bit far (not much though), in this case how do I tell him to choose a closer "parent"?

ebaauw, 2019-07-18:
For Xiaomi: Reset and re-pair the sensor.  You might also want to double-check the battery.  These sensors work very reliably, until the battery starts to fail.  This is not always reported correctly; any value under 80% or so might indicate trouble.

For any other sensor: power down the parent, trigger the sensor and it will find another one.

closed 2019-11-23 by stale[bot]