dresden-elektronik/deconz-rest-plugin #1676: IKEA Trådfri motion sensor detection

kabal00, 2019-07-13:
Hi all,

I have a strange behavior of my IKEA Trådfri motion sensor which has been perfectly synchronized with the deConz software (installed on raspberry pi 3B+). But after a motion has been detected all of my bulbs within deconz will be shown as triggered (ON), but they are definitely OFF. The bulbs are from OSRAM and IKEA and I am using Homebridge which works great, except this issue.

Does anybody know what I need to do? 

Thanks in advance

Cybersmurfen, 2019-07-14:
I got the same behaviour. All lights shows on when a motion is detected. 

It is the new waterproof detector - FW: 2.0.022 - the old detectors runs FW: 1.2.214

Cybersmurfen, 2019-07-14:
Well. I deleted the Trådfri sensor from deconz. If I don't reset the sensor it will keep triggering all lights.

avakarev, 2019-07-14:
I have issues with new model of motion control too.
The old model E1525 (2017) is fine
The new model E1745 (2019) cause deconz to report bulbs are on while they are off. Also deconz doesn't report motion events following the first "presense: true" till "presense: false".

ebaauw, 2019-07-14:
Maybe the new model sends different ZigBee commands/command parameters?  It’s definitely different hardware, no dials, no duration setting, but a 30% vs 100% brightness setting.  Do you have the sensor on Day or on Night setting?

avakarev, 2019-07-15:
@ebaauw Hard to say if it's on Day or on Night, there is no indicator anymore on the back. Now there is just a toggliing button, and I think it's supposed to be indicated by gateway/app what's the current settings. In my case, it's set by default, I didn't modify it. The older E1525 has default mode is set to "on Day". Here is how back side looks of E1745 now: 
[image]

avakarev, 2019-07-15:
@ebaauw This is how response of `GET /sensors/<id>` looks:
```js
{ config:
      { alert: 'none',
        battery: 87,
        delay: 180,
        duration: 60,
        on: true,
        reachable: true },
     ep: 1,
     etag: '32101bc3e3b2992cb8f7efd5e68cb04d',
     manufacturername: 'IKEA of Sweden',
     modelid: 'TRADFRI motion sensor',
     name: 'Bathroom Motion Sensor R',
     state:
      { dark: true,
        lastupdated: '2019-07-14T21:08:45',
        presence: false },
     swversion: '2.0.022',
     type: 'ZHAPresence',
     uniqueid: '14:b4:57:ff:fe:52:3c:3a-01-0006' }
```

Cybersmurfen, 2019-07-15:
@avakarev - if you push the toggle button once, the sensor shows if it is day or night with a small red indicator just next to the button. If you push it again it changes mode.

ebaauw, 2019-07-15:
> Hard to say if it's on Day or on Night, there is no indicator anymore on the back. Now there is just a toggliing button

The IKEA "manual" seems to suggest there's a LED that lights up.  Maybe I'm too optimistic reading it.

> This is how response of GET /sensors/<id> looks like:

Thanks.  Some observations:
- Unsurprisingly, it has the same `manufacturer name` and `model` (or the REST API plugin wouldn't have exposed it already).
- `config.delay` is consistent with the fixed duration of 3 minutes from the manual.
- What really worries me is the lack of a `config.group`, suggesting the sensor hasn't been bound to one.  The old model would select a random group on reset and send groupcast commands by default, but apparently this model sends a broadcast by default (which causes deCONZ to mark all lights as on).  But somehow the lights themselves don't turn on...
- `state.dark` is `true` which means that either:
  - The sensor is on Day mode;
  - The sensor is on Night mode, and it's actually dark enough to turn on the lights.

Can you try the following:
- In the GUI, bind the client _OnOff_ cluster of the motion sensor to a group.  Trigger the motion sensor and check that `config.group` is created.
- Run deCONZ with `--dbg-aps=2` and observe the logfile for something like:
  ```
  Jul 15 12:09:00 pi1 deCONZ[1121]: 12:09:00:346 APS-DATA.indication srcAddr: 0xe9b9, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -55
  Jul 15 12:09:00 pi1 deCONZ[1121]: 12:09:00:346 	asdu: 0162420058020000
  Jul 15 12:09:00 pi1 deCONZ[1121]: 12:09:00:347 button 1002 On with timed off
  ```
  The `asdu` means the following:
  ```
  0162420058020000
      | | |   + 0x0000: _Off Wait Time_: 0.0s
      | | + 0x0258: _Timed Off_: 60.0s (i.e. `state.delay` ==> 60)
      | + 00: _Accept only when on_ is not set (i.e. `state.dark` ==> `true`)
      + command: 0x42: _On With Timed Off_
  ```
- Add a light to the group, and see if the motion sensor will control it.  Even if the light won't turn on, the REST API plugin should only mark the light(s) in the group as on, instead of all lights.

tubalainen, 2019-07-17:
> > Hard to say if it's on Day or on Night, there is no indicator anymore on the back. Now there is just a toggliing button
> 
> The IKEA "manual" seems to suggest there's a LED that lights up. Maybe I'm too optimistic reading it.
> 
> > This is how response of GET /sensors/ looks like:
> 
> Thanks. Some observations:
> 
> * Unsurprisingly, it has the same `manufacturer name` and `model` (or the REST API plugin wouldn't have exposed it already).
> * `config.delay` is consistent with the fixed duration of 3 minutes from the manual.
> * What really worries me is the lack of a `config.group`, suggesting the sensor hasn't been bound to one.  The old model would select a random group on reset and send groupcast commands by default, but apparently this model sends a broadcast by default (which causes deCONZ to mark all lights as on).  But somehow the lights themselves don't turn on...
> * `state.dark` is `true` which means that either:
>   
>   * The sensor is on Day mode;
>   * The sensor is on Night mode, and it's actually dark enough to turn on the lights.
> 
> Can you try the following:
> 
> * In the GUI, bind the client _OnOff_ cluster of the motion sensor to a group.  Trigger the motion sensor and check that `config.group` is created.
> * Run deCONZ with `--dbg-aps=2` and observe the logfile for something like:
>   ```
>   Jul 15 12:09:00 pi1 deCONZ[1121]: 12:09:00:346 APS-DATA.indication srcAddr: 0xe9b9, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -55
>   Jul 15 12:09:00 pi1 deCONZ[1121]: 12:09:00:346 	asdu: 0162420058020000
>   Jul 15 12:09:00 pi1 deCONZ[1121]: 12:09:00:347 button 1002 On with timed off
>   ```
>   
>   
>   The `asdu` means the following:
>   ```
>   0162420058020000
>       | | |   + 0x0000: _Off Wait Time_: 0.0s
>       | | + 0x0258: _Timed Off_: 60.0s (i.e. `state.delay` ==> 60)
>       | + 00: _Accept only when on_ is not set (i.e. `state.dark` ==> `true`)
>       + command: 0x42: _On With Timed Off_
>   ```
> * Add a light to the group, and see if the motion sensor will control it.  Even if the light won't turn on, the REST API plugin should only mark the light(s) in the group as on, instead of all lights.

Hi! Can confirm that the new IKEA sensor is behaving oddly. It does not create the config.group and "ghosts" all lights in deconz to "ON" on motion detected. Ive tried the @ebaauw way described above with no success.

How ever, can confirm that the sensor is VASTLY more sensitive than the old one. Hoping for a solution. <3

avakarev, 2019-07-19:
@ebaauw 
> The IKEA "manual" seems to suggest there's a LED that lights up. Maybe I'm too optimistic reading it.
I can confirm the back side shows the state and it was on Day mode.

Unfortunately I don't have the access to conbee stick / deconz at the moment so can't provide the logfile / debug output. Hopefully some users of E1745 sensors can help with that.

@ebaauw Thank you for your involvement and attempt to help!

Ergerbolt, 2019-07-24:
Hi there, 

same problem for me. Exactly same behavior.

JustAnotherUserName87, 2019-07-30:
Hi,

I have the same problem. Unfortunately I only have the newer sensor.

ebaauw, 2019-08-11:
Made my semi-annual trip to IKEA this weekend, because I needed some new batteries and I long wanted to try the outlet with on/off switch.  Unfortunately, the motion sensors were sold out completely, both old and new model. So I got a SYMFONISK instead...

Still waiting on the FYRTUR, so I might actually visit them again by the end of the month.

Michaelnorge, 2019-08-11:
I have arround 20 Trådfri plug sockets.
I synchronized two of the new motion sensors from Trådfri and got the same problems like described here.

Even after deleting the motion sensors the items keep turning on my lights.
I needed to reset them.

After that almost all of my plug sockets were offline. I needed to resynch them.

I don´t think that there is someone who will fix this problem. 
The first entry here is about a month ago and still nothing happend...
Bummer! :-(

ebaauw, 2019-08-11:
> Even after deleting the motion sensors the items keep turning on my lights.

That is to be expected.  The motion sensor sends a broadcast command to turn on all lights.  This works without any interference from deCONZ, so removing the motion sensor from deCONZ won't change this.  You need to remove the motion sensor from the network (by resetting the motion sensor).

> I needed to reset them.
> After that almost all of my plug sockets were offline. I needed to resynch them.

Again, this is to be expected.  When you reset the sockets, they are removed from the network.

> I don´t think that there is someone who will fix this problem.
> The first entry here is about a month ago and still nothing happend...

Becasue no-one has provided the info we need to fix it.  To gather the info myself, I need the device...

tubalainen, 2019-08-11:
> That is to be expected. The motion sensor sends a broadcast command to turn on all lights. This works without any interference from deCONZ, so removing the motion sensor from deCONZ won't change [...]

I can provide the info needed (I hope...), just tell me what you need. Screendumps of info, debug?

//T

ebaauw, 2019-08-11:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1676#issuecomment-511347951

triumvirrr, 2019-08-12:
@ebaauw 
I tried the suggested steps.

1) I am sorry, but I did not entirely understand the first step, what to do exactly and where to look for config.group. Probably I need the GUI for this? My Raspberry runs headless at the moment.

2) My sensor sends the folllowing command when motion is detected:
09:20:11:389 APS-DATA.indication srcAddr: 0x98be, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 103, rssi: -78
09:20:11:389 	asdu: 011a420008070000

3) Binding the sensor to a group of lights does not change its behaviour, still all lights are ghost-on after a movement.

I will happily make further tests, especially test 1, with your guidance.

ebaauw, 2019-08-12:
Ad 2.  I assume the sensor is on Day mode, and it’s light enough?  This is the expected _On with timed off_ command, with _Accept only when On_ and a 3-minute timed off.  Same as the old motion sensor, I don’t understand why the lights don’t respond.  Must be something special with the group.  I’m not sure if that can be seen in the log, I probably need ZigBee sniffing output.

How did you bind the sensor for 3 if you don’t understand 1?  Yes, you need the GUI for this.

triumvirrr, 2019-08-12:
Yes, the example was in day mode with sufficient light.

Coming back to your test #1:
I do not understand what you mean with "the client OnOff cluster of the motion sensor" and I do not know where to look for config.group. Could you assist me with some hints?
Test  #3 could be done via the web interface.

Just to be clear:
The issue is not, that the lights do not go on.
The issue is, that ALL the lights, regardless of their configuration regarding the motion sensor (in my case mostly a bunch of Osram Smart+ Plugs) are marked as on, although they (probably) are not. In fact, they keep the physical state as before but are shown to be on for 1 minute. This false state is then also passed on to Openhab which causes quite some mess.

I have collected further logs, perhaps they are useful:
Night mode:
16:48:41:728 APS-DATA.indication srcAddr: 0x98be, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 247, rssi: -60
16:48:41:728 	asdu: 0130420108070000
...
16:49:49:015 aps request id: 214 prf: 0x0000 cl: 0x0031 timeout (confirmed: 1) to 0x000D6FFFFE45ADBA (0x98BE)
16:49:49:094 aps request id: 214 finished, erase from queue

Night mode, light setting "dimmed" (instead of 100%) in the motion sensor:
16:55:04:344 APS-DATA.indication srcAddr: 0x98be, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 111, rssi: -77
16:55:04:344 	asdu: 0132420108070000
...
16:56:13:015 aps request id: 228 prf: 0x0000 cl: 0x0031 timeout (confirmed: 1) to 0x000D6FFFFE45ADBA (0x98BE)
16:56:13:095 aps request id: 228 finished, erase from queue

Additional information:
The sensor is shown with "power source: main" and has lost 13% of its battery in one day. I do not know if this is helpful or if it correlates to the problem.

ebaauw, 2019-08-12:
> I do not understand what you mean with "the client OnOff cluster of the motion sensor" and I do not know where to look for config.group. Could you assist me with some hints?

In the deCONZ GUI, you see the device's end points and clusters: blue for the server clusters and grey for the client clusters.  You need to bind the client _On/Off_ cluster 0x0006 to a group, using the _Bind Dropbox_ panel.  This is what it would look like for the old model:
[image].

Make sure to wave in front of the sensor to wake it, and press _Bind_.  Wave again after half a minute or so to make the sensor (hopefully) send a command to the bound group, and the REST API should show `config.group` when GETting the `/sensors` resource.  For the old model, it would be something like:
```
$ ph get /sensors/9
{
  "config": {
    "alert": "none",
    "battery": 34,
    "delay": 60,
    "duration": 60,
    "group": "3797",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "45ba72f1e5fa02ab0e4cb28e7f83dff7",
  "manufacturername": "IKEA of Sweden",
  "modelid": "TRADFRI motion sensor",
  "name": "TRADFRI motion sensor ",
  "state": {
    "dark": true,
    "lastupdated": "2019-08-12T18:25:29",
    "presence": false
  },
  "swversion": "1.2.214",
  "type": "ZHAPresence",
  "uniqueid": "00:0b:57:ff:fe:22:cf:5f-01-0006"
}
```
> The issue is not, that the lights do not go on.
The issue is, that ALL the lights, regardless of their configuration regarding the motion sensor (in my case mostly a bunch of Osram Smart+ Plugs) are marked as on, although they (probably) are not.

The issue is that the motion sensor sends a command that deCONZ thinks turns on all lights, but that the lights seem to ignore.  This first part makes me thing it's a kind of broadcast command, which we hope to change to a groupcast by binding the sensor to a group.  The second part is really interesting, but might not be an issue if don't need the sensor to control the lights directly, without using the gateway.

> In fact, they keep the physical state as before but are shown to be on for 1 minute

I suppose until next polled by deCONZ, so the timing might differ per light and will depend on the number of lights in your network.

> I have collected further logs, perhaps they are useful:

Cool.  Yes they are.  I think I managed to confuse myself once again.  ASDU xxxx 42 00 0807 0000 is turning the lights on unconditionally (i.e. when the sensor in on _Day_, or when the sensor is on _Night_ and it's dark); ASDU xxxx 42 01 0807 0000 is prolonging the on duration when the lights were already on, or does nothing when the lights were off (i.e. sensor is on _Night_, but it's not dark).

The really interesting part is that the same payload seems to be used for the _Dimmed_ setting.  Either the sensor sends another command after this one (I'd expect cluster 0x0008 or maybe 0x0005), or it uses another broadcast address (not visible in the log).  I'm afraid this might be another IKEA special, similar to the way the remote handles changing colour or colour temperature.

The timeouts on the ZDP (profile 0x0000) requests are unrelated.  This is deCONZ trying (and failing) to get some network info from the sensor.

ebaauw, 2019-08-12:
Oh, and it would help to have the GUI screenshots of the new model.  In particular: is it now an HA / Zigbee 3 device, like the On/Off switch, or still a ZLL device; and does it have an 0xFC7C cluster.

BeamMeUpTo, 2019-08-13:
Hi all of you ;),

I will try to send additional information - so here first is the screenshot of the new ikea motion sensor from the deCONZ gui:

[image]

The Binding Dropbox is empty!

I would like to provide additional info about
`ph get /sensors/9`

but I have no clue what "ph" is! Please enlighten me!

Beam

ebaauw, 2019-08-13:
> here first is the screenshot of the new ikea motion sensor from the deCONZ gui:

Thanks.  Some observations:
- This is yet another mac address prefix, which hasn't been whitelisted in the REST API code.  Do you actually see the sensor in the API and/or Phoscon?
- It looks a lot like the on/off switch, including 0xFC7C IKEA cluster.  I can only assume the Trådfri hub interacts with the cluster to configure the sensor;
-  0x0008 _Level Control_ client cluster.  This strengthens my suspicion that the sensor might issue _Level Control_ commands when _Dimmed_ is set.

> The Binding Dropbox is empty!

Yes, no need to drag the 0x0008 cluster to the _Source_ drop box and fill in the group ID manually.

> I have no clue what "ph" is

Just a liitle tool I wrote to interact with the Hue/deCONZ API from the command line.  It comes bundled with homebridge-hue.  You can use `curl` instead, but then you have to specify the full URI (incl. host and username/apikey) and format the JSON yourself.

triumvirrr, 2019-08-13:
Thanks, Eric.

Binding the sensor via GUI binding dropbox to an (in my case) empty group did the trick. No Ghost-Ons anymore. I also tested with a group with lights in it and they are switched as expected. And only the lights in the group are switched.

Side effect: This group is then not visible via Web Interface anymore...

The sensor is normally visible within all GUIs and the REST API.

Here is the sensor output via REST after binding it to group 6:
```
{
"config": {
"alert": "none",
"battery": null,
"delay": 180,
"duration": 60,
"group": "6",
"on": true,
"reachable": true
},
"ep": 1,
"etag": "26704aff65b268ca9d2b4aa67fc75963",
"manufacturername": "IKEA of Sweden",
"modelid": "TRADFRI motion sensor",
"name": "TRÅDFRI Bewegungsmelder",
"state": {
"dark": true,
"lastupdated": "2019-08-13T10:06:58",
"presence": true
},
"swversion": "2.0.022",
"type": "ZHAPresence",
"uniqueid": "00:0d:6f:ff:fe:45:ad:ba-01-0006"
}
```

BeamMeUpTo, 2019-08-13:
Yes,

my sensor Sz.Motion is visible in phoscon!

After the tip with LevelControl Cluster I changed the GroupID to 0x0006 which in my case does not exist ;) and seems to be a workaround for me.

Reason:
I noticed in FHEM that everytime the motion sensor detects a motion HUEGroup0 got triggered in Event Monitor of FHEM. Which results in pseudo on-switching of Osram Smart+ Plugs. (belonging to HUEGroup0!)

This effect is now gone. In fhem I can use the IKEA Motion Sensor without any pseudo switching!

Beam

LJSven, 2019-08-13:
I have the same behaivor - what do i have to do to fix it? I´m using ioBroker, Phoscon and deconz

BeamMeUpTo, 2019-08-13:
Workaround:

Connect IKEA Motion Sensor to phoscon app as normal.

Start deCONZ GUI.

Choose sensor and activate the right dot! Register panel opens.

Activate the Binding Dropbox Window: Menu --> Panels --> Bind Dropbox

Now drag "0008 LevelControl" to Source in the Bind Dropbox Window.

Change the GroupID to a not existing one try 0x0010!

Press Bind!

[image]

ebaauw, 2019-08-13:
Cool!  So we need the REST API plugin to pick a random unused group and create the binding upon pairing the sensor.

> Now drag "0008 LevelControl" to Source in the Bind Dropbox Window.

That's actually a lucky typo on my part - I meant the 0x0006 (_On/Off_) cluster.  Although I've seen other devices that use the binding for one cluster for the other clusters as well.

> Press Bind!

Make sure the sensor is awake, or you might get a timeout.

> my sensor Sz.Motion is visible in phoscon!

@BeamMeUpTo, I'm happy for you, but I don't understand how this is possible.  Can you please double-check that the `uniqueid` indeed starts with `14:b4:57` (and not with `00:0d:6f` like @triumvirrr 's)?

LJSven, 2019-08-13:
I don´t have the GUI installed. Is there any other way?

BeamMeUpTo, 2019-08-13:
@ebaauw 
So, I double-checked ;)

I bought two of those sensors last weekend, and both of them have the 14:b4:57. and the known problem of pseudo-switching.

[image]

I tried to drag "0006 on/off" to source and afterwards checked the GroupID for the other cluster - it is everywhere GroupID 0x0006

@LJSven:
I don't know if there is another way to change GroupID - Eric?

But I had to install the deconzGUI on my raspi, too and use it with vncviewer, because of headless. Not nice but it works ;)

ebaauw, 2019-08-13:
> afterwards checked the GroupID for the other cluster - it is everywhere GroupID 0x0006

You the group ID displayed is the input for new bindings.  There's no way to retrieve (the group ID from) an existing binding in the GUI.  The only way to check is to trigger the sensor and check what the API reports for `config.group`.

> I don't know if there is another way to change GroupID - Eric?

You need to create a binding by sending the corresponding ZigBee command to the sensor or switch.  The only way to do that manually in deCONZ is through the GUI.  The REST API does this automatically for whitelisted devices, so normally you wouldn't bother.

If you want more control, you could issue the ZigBee bind command and query the binding table through the deCONZ CLI plugin, see https://github.com/ma-ca/deconz-cli-plugin.  But that's not for the faint-hearted: keep your ZigBee documentation and binary/hex - decimal converter at hand.

ebaauw, 2019-08-13:
> I don't understand how this is possible.

Use the Source, Luke:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/254f1a8e530c635f9d430dfaddf45d9e87d997b7/de_web_plugin.cpp#L6720-L6721
So the REST API code checks for matching manufacturer code OR matching mac address.
@manup, I think this is a bug; shouldn't both match (i.e. L6720 should end in `&&` instead of `||`)?

Ergerbolt, 2019-08-13:
Hi, i think you all have a solution for the Problem. Very nice! Thank you! If im Right, i Need dCONZ gui on my pi. How can i install this program on hassio? :)

ebaauw, 2019-08-13:
The deCONZ GUI is already installed, as it's integral part of deCONZ.  Depending on how you start deCONZ, it shows the GUI (by default) or not (`-platform minimal` option).  If you use `systemctl` to start deCONZ as a service (normal Raspberry Pi installation), deCONZ provides a `deconz.service` and a `deconz-gui.service`.  Make sure to enable/start only one of these and disable the other.

Of course you need to start the graphical environment of your OS to run the GUI.  No clue how to do that on hassio.

LJSven, 2019-08-13:
Can i do the changes on another PI or do i have to do it on the same where phoscon is running?

ebaauw, 2019-08-13:
Phoscon is a web app that runs in your browser.  In connects to deCONZ running on your Pi over the REST API.  The GUI runs on that same Pi, as it's integral part of deCONZ.

You can run deCONZ with the GUI enabled on a headless Raspberry Pi, using VNC to access the graphical environment remotely.  I actually use macOS screen sharing to interact with my Raspberry Pi-s, instead of the VNC client.  And Mocha VNC on my iPad.

triumvirrr, 2019-08-13:
Can I sniff the command when I just have one RaspBee hardware which is used for deconz, obviously? If so, I will gladly do it - perhaps with one or two hints on which tool to use :-)

LJSven, 2019-08-13:
When i connect to my Tinkerboard via VNC it shows the following message ->

> Xsession: unable to start X session --- no "/root/.xsession" file, no 
"/root/.Xsession" file, no session managers, no window managers, and no 
terminal emulators found; aborting.

ebaauw, 2019-08-13:
> Can I sniff the command when I just have one RaspBee hardware which is used for deconz, obviously?

No, you need a second ZigBee radio device, typically with sniffing firmware.  I use a ConBee with ZShark, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/405, on a separate Pi.

> When i connect to my Tinkerboard via VNC it shows the following message

Looks like the graphical environment hasn't been started.  I would guess it's installed, as the VNC client seems to find a VNC server on the Tinkerboard to talk to.

No clue how to start the graphical environment on a Tinkerboard; for the Raspberry Pi is through `sudo raspi-config` (_3 Boot Options_ and then _B1 Desktop / CLI_).  Same to enable the VNC server (_5 Interfacing Options_ and then _P3 VNC_).

Ergerbolt, 2019-08-13:
i got it. Now i bonded 0008 Cluster to Gruup 0x0010. But now my Group "Bath" is away :-D

BeamMeUpTo, 2019-08-13:
0x0010 was an example, which was sufficent for my setup.

If you have more groups than 10 - you have to choose a higher value! ;)

Ergerbolt, 2019-08-13:
Ok, hmm, i will try. When i use 0x0006 some lights - from group 6 - are switched on. 
Do you know how i can get my groups in the phoscon app back?

BeamMeUpTo, 2019-08-13:
0x0006 is lower than 0x0010, so it is clear, that you switch on another group, if you had already difficulties with 0x0010 - try 0x0100!

Ergerbolt, 2019-08-13:
Sure, you are right, i just wanted to tell, that ... hmm i don't know what i wanted to say. Wich cluster do i have to change? The "Level control" or the "on/off" cluster? I'm not sure because there are two descriptions in the earlier threads?

BeamMeUpTo, 2019-08-13:
on/off cluster is the save way

Cybersmurfen, 2019-08-13:
Well. I'm connected to homeassistant as well and are using the event listener to trigger automations.

I decided to add a new motion sensor and except for all above, it looks like it doesn't come out any events from deconz at all while the sensor is connected.

And yes, there is 3 buttons on the back. One toggler for day/night, one toggler for wide detection and one link button. When toggling the button, a small red light indicates which mode is selected.

The new motion sensor is identified as an old one, and the firmware is 2.0.22

In the phoscon app, you can add it to a motion control and it is working as expected, but all other lights are "ghosted on".

And to remove it, just remove it from the app and then reset the sensor.

LJSven, 2019-08-14:
It looks like, if you switch the day & night button to "night" the problem is not happening.

BeamMeUpTo, 2019-08-14:
I played around with REST API so I could really verify my groupID Settings: (In my case it made no difference if I write 0x0006 to the wrong Level Cluster or into the right ON/OFF Cluster)

Here is the output from both Sensors (FYI Erik):

`"6": {
    "config": {
      "alert": "none",
      "battery": 100,
      "delay": 180,
      "duration": 60,
      "group": "6",
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "17ce7f393dd778ad96aa7a3d5336eba5",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI motion sensor",
    "name": "TRÅDFRI Bewegungsmelder",
    "state": {
      "dark": true,
      "lastupdated": "2019-08-14T06:10:17",
      "presence": false
    },
    "swversion": "2.0.022",
    "type": "ZHAPresence",
    "uniqueid": "14:b4:57:ff:fe:3e:b1:47-01-0006"
  },
  
    

"9": {
    "config": {
      "alert": "none",
      "battery": 100,
      "delay": 180,
      "duration": 60,
      "group": "6",
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "d84a07018017428aea5f33be24c7827b",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI motion sensor",
    "name": "Sz.Motion",
    "state": {
      "dark": true,
      "lastupdated": "2019-08-13T20:25:46",
      "presence": false
    },
    "swversion": "2.0.022",
    "type": "ZHAPresence",
    "uniqueid": "14:b4:57:ff:fe:3e:ad:0d-01-0006"
`

tubalainen, 2019-08-14:
Got it to work! Sorry @manup didnt understand "drag and drop to source". WIth some trial and error I got the new IKEA Motion sensor to work via the REST API to Home Assistant!

This were the steps for the deconz REST-API:
0.) You need to reach the GUI. I use Marthoc´s docker (https://github.com/marthoc/docker-deconz) and have enabled VNC server in the docker. I then use the vncviewer application windows to connect to the Deconz GUI. 
1.) Add the sensor just like normal in Phoscon
1.5) On the back of the sensor, toggle the night/day mode so that the crested moon lits up
2.) Connect using VNC (in my case) to your deconz instance
3.) Go to "Panels - Bind dropbox"
4.) Locate your sensor amongst the rest of your nodes in the GUI
5.) Push the far right of the two grey circles on your sensor note
6.) Drag and drop the "0004 Groups" cluster to SOURCE in the "Bind dropbox" panel
7.) Select "Group" and enter the value 0x0100
8,) Trig the sensor my moving (so it is awake) and push "bind"
9.) Restart deconz and home assistant and you should be all good to go.
[image]
[image]
[image]
[image]

Cybersmurfen, 2019-08-14:
I can confirm that tubalainens solution works like a charm.

Tackar!

ahd71, 2019-08-14:
I also confirm that the workaround with binding a new group worked for me! Thanks all.

Some clarification (0x0010 is likely hex 10 which is 16), but perhaps most improtant I added a last step of "walk in front of sensor" after pressing the bind button. It was NOT necessary to restart deconz or home assistant for me after the binding as hinted in some other guide.

- Connect IKEA Motion Sensor to phoscon app as normal.
- Start deCONZ GUI.
- Choose sensor and activate the right dot! Register panel opens.
- Activate the Binding Dropbox Window: Menu --> Panels --> Bind Dropbox
- Now drag "0006 On/Off" to Source in the Bind Dropbox Window.
- Change the GroupID to a not existing one try 0x0010! (Example!!! Only use if you have less than 0x0010 Groups!)
- Press Bind!
- Walk in front of the sensor - you should see a confirm message in the bind dialog if it works - otherwise a timeout message

Cybersmurfen, 2019-08-14:
@adh71 - "0006 On/Off" ?
tubalainen stated "0004 Groups" and that worked for me.
Typo or did both work?

ahd71, 2019-08-15:
I tried only 0006

ahd71, 2019-08-15:
I tried only 0006

Ergerbolt, 2019-08-15:
I think, its both working fine, because i took 0008. BeamMeUpTo already confirmed this as well.

manup, 2019-08-15:
> Use the Source, Luke:
> https://github.com/dresden-elektronik/deconz-rest-plugin/blob/254f1a8e530c635f9d430dfaddf45d9e87d997b7/de_web_plugin.cpp#L6720-L6721
>
>
> So the REST API code checks for matching [...]

Should be fine, first branch checks if the node descriptor provides a valid manufacturer code and as alternative the mac prefix based checks are applied, therefore the `||`.

Our new IKEA sensors should arrive here soon so will do some more tests to automate the current manual configuration.

ebaauw, 2019-08-15:
@manup, Welcome back!

I meant the `||` at the end of line 6720, not the one on line 6721.  I think the logic should be:
```
!isNull AND manufacturerCode AND (mac OR checkVendorMac)
```
instead of
```
(!isNull AND manufacturerCode) OR (mac OR checkVendorMac)
```

martinkrulltott, 2019-08-22:
Hey, I'm experiencing this issue as well, running on a Raspberry Pi 3B+ and the deconz docker image (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1767). It looks like you guys are accessing a GUI, but I'm running this headless in Docker, anyway to fix this through the console?

Ergerbolt, 2019-08-22:
I think you can use the vnc device protokoll

[58 more replies]

closed 2019-09-15 by manup