dresden-elektronik/deconz-rest-plugin #1680: deCONZ fails to start after installation

sambal0, 2019-07-16:
Hey!
I have installed the latest deb on my Raspberry which is used successfully as a Homebridge for a couple of months. Wanted to add an OSRAM Smart+ plug today through the homebridge-hue plugin. That is why I need deCONZ. Unfortunately none of the discussed or solved issued helped me to get deCONZ started:

`pi@raspberrypi:~ $ sudo systemctl status deconz-gui
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/etc/systemd/system/deconz-gui.service; enabled; vendor preset: enabled)
   Active: failed (Result: signal) since Wed 2019-07-17 01:44:56 CEST; 6s ago
  Process: 2911 ExecStart=/usr/bin/deCONZ --http-port=80 (code=killed, signal=ABRT)
 Main PID: 2911 (code=killed, signal=ABRT)

Jul 17 01:44:55 raspberrypi systemd[1]: deconz-gui.service: Failed with result 'signal'.
Jul 17 01:44:56 raspberrypi systemd[1]: deconz-gui.service: Service hold-off time over, scheduling restart.
Jul 17 01:44:56 raspberrypi systemd[1]: Stopped deCONZ: ZigBee gateway -- GUI/REST API.
Jul 17 01:44:56 raspberrypi systemd[1]: deconz-gui.service: Start request repeated too quickly.
Jul 17 01:44:56 raspberrypi systemd[1]: Failed to start deCONZ: ZigBee gateway -- GUI/REST API.
Jul 17 01:44:56 raspberrypi systemd[1]: deconz-gui.service: Unit entered failed state.
Jul 17 01:44:56 raspberrypi systemd[1]: deconz-gui.service: Failed with result 'signal'.`

Downgraded to 2.05.21 so far and tried to `stop`, `start`, `enable` and `restart` manually `deconz`, `deconz-gui` and `deconz-gui.service`. A reboot of the Raspberry did not helped either.
Can anybody help me?

Smanar, 2019-07-17:
Have you tried using the raspberry GUI instead of command line ?

sambal0, 2019-07-18:
Not yet. You mean through the command line within the Raspberry GUI instead of using the command line in Terminal on MacOS, right?

Smanar, 2019-07-18:
Without all command line, just using the GUI, go in menu then on "programmation" and "deconz". You are using the GUI version so it can be possible.

sambal0, 2019-07-18:
Thanks for your help! After a lot of other discussions I found out that my Raspberry Stretch Lite had no GUI installed. On this way I updated my Raspberry and crashed bluetooth by the way...
Anyway after installing all the missing files I can see the deCONZ app itself. Unfortunately, it is not working properly. A firmware update is needed which will not be installed. But that is a another issue. Thank you!

closed 2019-07-18 by sambal0