dresden-elektronik/deconz-rest-plugin #1683: Save Toggle Power State

andjandre, 2019-07-17:
I control my lights via Homekit and via two physical buttons (Xiaomi Button and Ikea Tradfri Remote). 
I set both power buttons to toggle the lights. This works fine but when I turn off the lights via Homekit, then I need to press the toggle button twice to turn the lights on. It seems like the toggle state won't realize the change caused by Homekit. 
What can I do to get this working?

Smanar, 2019-07-17:
Yours buttons works directly with zigbee network (or through deconz), or you use automation from Homekit (to say it it more clearly, if you turn off homekit, yours button are still working ) ?

andjandre, 2019-07-17:
Thanks for the quick reply. It works directly via zigbee (configured via phoscon). I wanted to have a fast response as possible

Smanar, 2019-07-17:
I don't have your hardware so I can't help you a lot, but it's realy strange for me.
If you have a GUI version.
- Go on cluster 0006 "on/off"
- On the botton you have the attribute, take a look on this one called "on/off" (bool).
- If you use the homekit toogle fonction this value go from off to on (and on to off) ?
- If you use the "toogle fonction from deconz (same cluster info panel), it works or you need too 2 press ?

closed 2019-11-21 by stale[bot]