dresden-elektronik/deconz-rest-plugin #1686: Request for 'Level Control Switch

udoewich, 2019-07-18:


udoewich, 2019-07-24:
I'm working on adding support for this to the code.  A couple of questions:

Is it possible to override the manufacturer string, and if so, where in the source code should that be done? I'm asking since the device I'm working with reports the manufacturer in the original Chinese characters (欧瑞博) - would be nice to replace it with the westernized version (Orvibo).

After making some code changes I've now managed to add this in-wall switch as a dimmable light, but I've noticed some odd behavior in Phoscon:

When I click on the light bulb, the light correctly switches off, but the brightness slider is also set to zero. This should not happen as brightness should be independent of the on/off state.  If I now click on the bulb again to turn the light back on, the physical lights turn back on, but the brightness slider remains at 0.  Is it possible to decouple the on/off from the brightness?

Smanar, 2019-07-24:
To change manufacturer string take a look on de_web_plugin.cpp on fonction void DeRestPluginPrivate::addLightNode(const deCONZ::Node *node)

For the second problem, I have no idea, I have make tries with bulbs and they works same, on phoscon when I set the bulb off, the slider go to 0, but when I turn it on again, the slider go to the previous value. I have check on deconz, even the slider go to 0, the previous values is still in still in the attribute "Current level"

udoewich, 2019-07-25:
Thanks, I'll give that a try.

Managed to suppress the color cluster today that the device advertises.  Doesn't make sense for an in-wall dimmer, since it can only change the brightness of the connected load.

So now settings for this "light" show up properly in scenes:

`        {
            "bri": 191,
            "id": "12",
            "on": true,
            "transitiontime": 10
        }`

However, recalling this scene via Phoscon does nothing: the lights do not turn on.

Any ideas on how to 'debug' this?

udoewich, 2019-07-26:
@manup I'm having a bit of trouble with my ORVIBO in-wall dimmer and switch: both don't seem to respond at all when recalling a scene.  Setting on/off state and brightness (for the dimmer) via Phoscon controls and the API work just fine.

In deCONZ, if I try to use the read Attributes button on the Scenes cluster, deCONZ crashes.  What do I have to do to get a better idea why the crash happens?

udoewich, 2019-07-27:
Did some more testing:

looks like ORVIBO's implementation of Scenes is broken, or at least doesn't work the way it's supposed to.  Adding a scene works in the sense that the scene is created: doing a query with View Scene shows it exists and has stored the transition time. However, Recall does not restore either the on/off state or the brightness level that was in use when the scene was added, or when Store is used.

I noticed they "auto-create" two scenes with group id 0. Recalling these two works. Scene 1 turns the dimmer on, to the last used brightness level. Scene 2 turns the dimmer off.  Could very well be that that's how they store the last brightness level when you physically use the switch.

Trying to read the Attributes of the Scenes cluster still causes deCONZ to crash; confirmed with a unmodified v2.05.66 installation under Windows. I suspect what is returned is not in the format deCONZ expects, but I don't know how I can confirm that.

Is it possible to suppress the Scenes cluster for these devices? Would the REST API and Phoscon use direct control of on/off and brightness in that case?