dresden-elektronik/deconz-rest-plugin #1687: Feature: Adding milliseconds to the `lastupdated` field.

vandalon, 2019-07-18:
Adding milliseconds to the `lastupdated` field would be nice to have. I use automation in node-red that count the amount of presses of a button in $time (mostly 1 second) for switches that don't support double click them self. In node-red I check the `lastupdated` field to make sure that nothing happens if the switch just sends out a status update. (some switches do, some don't :)).

vandalon, 2019-11-15:
Still wanted :)

ebaauw, 2020-03-14:
> Adding milliseconds to the `lastupdated` field would be nice to have.

Yeah, I'd think so.  Not sure API clients (incl. homebridge-hue) would be able to handle it without changes.  The ISO 8601 date format supports it, so I suppose they (we) should shut up and get their (our) act together ;-).

> I use automation in node-red that count the amount of presses of a button in $time (mostly 1 second) for switches that don't support double click them self.

Imho, that's looking for trouble, especially if you want Hue bridge compatibility.  I would tell you to get a switch that supports double-press natively, but then again, this info might be hard to get before buying the switch.

>[...] if the switch just sends out a status update. (some switches do, some don't :)).

That sounds wrong to me.  I don't know of any wireless switch that send periodic updates that would cause `buttonevent` or `lastupdated` to change.  Afaik, periodic updates are only used only to update `config.battery`, and maybe `config.temperature` for the Xiaomi switches.  Neither should set `state.lastupdated`.

ebaauw, 2020-03-15:
I managed to display `lastupdated` with millisecond resolution.  The `ResourceItem` value only maintains a resolution of seconds, but the `lastSet()` value of the item is in milliseconds ;-).  homebridge-hue happily accepts the value.  In fact, I think we don't need any value at all - `lastSet()` does just fine.

There's two other changes I would want us to consider, please provide input:
- When there's no valid date, the value is set to `"none"`.  I think this should be `null` instead;
- Would it make sense to add a `Z` to the value to indicate Zulu-time (UTC) explicitly.

gummigroda, 2020-03-15:
I really need the `Z` to be added to the value, regardless of milliseconds or not.

As of now, using Date.Parse() on the value, [JavaScript](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Date/parse#Examples) assume it's local time because of the missing TZ info.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

ebaauw, 2020-06-06:
Still open whether / when to add the `Z` to the date/time fields.