dresden-elektronik/deconz-rest-plugin #1691: No Pairing without powercycle

thecem, 2019-07-18:
RPI3B+, Raspian Strech Latest, Deconz minimal latest .66, Raspbee.

It is only the first ca 20 min after  powercycle posible to pair devices.

When I powercycle the raspbee it easy to pair devices after a while it is not possible anymore.

Thanks for help!

ebaauw, 2019-07-18:
As a security measure, most lights will only pair or reset shortly after they've been power cycled.  Power cycling the RaspBee shouldn't do anything for pairing.

thecem, 2019-07-19:
This is the point! 
I have to powercycle the RPI (RaspBee Module)!
The resetet devices (5 sec push) will pair immediately after pushing the pair button (short push).
After a wile 20min-1h it is not longer possible to pair, regardless which device.
When I make a backup, reset the configuration of RaspBee Phoscon GW and load the backup in again (without a powercycle), there is a new timeframe where I able to open the network via web and add devices, the device will pair immediately.
What i didn´t try by now is how long a blank new configuration after resetting the GW will allow me to pair devices...

thecem, 2019-07-19:
I tryed ..... 21 devices ...
I reset the gateway, pair 18 battery devices (xiaomi). After this I have to powercycle the raspi, to be able to pair. After a while i'm again not able to pair.
It seems to be that the web does not open the network.

Is it possible to log this if the network is open or not?

thecem, 2019-07-20:
additional when I'm powercycle the RaspBee (0x26330500) it is also for up to few minutes possible to pairing devices (permit join):

sudo systemctl stop deconz
sudo GCFFlasher_internal -r
sudo systemctl start deconz

I could send you a backup if you would like to analyse, or even some logs. thanks in advance....

thecem, 2019-07-30:
In between I testet it with ConBee 1 and same result, after a while of ca. 20 min not able to join devices to the network...

How can I analyse the reason behind this? 

If someone need logs I will provide ....

Thanks a lot!

Smanar, 2019-07-30:
I m realy curious, no other issues like this one ?

thecem, 2019-08-06:
Nobody can help here? @Smanar: Yes no other issues. Stable network without any interrupts. With a powercycle of the whole rpi or a 

sudo systemctl stop deconz
sudo GCFFlasher_internal -r
sudo systemctl start deconz

deconz will be able to join devices...
I can´t see any reason behind this and if someone has an idea who to trace or debug this please help.....

Smanar, 2019-08-06:
It's holiday perdiod, there is lot of issues without answer ATM.

thecem, 2019-08-10:
news: I bought a ConBee II (264A0700).
I disconectet the RaspBee Module and connected the ConBee II to the rpi3b USB Port (ACM0) and reload the backup. After 24h (with complete same setup/software/config/ postioning etc.) Phoscon will pair (join) devices! (-: Same after 3 days 😃 !!!

We rember with raspbee and conbee (1) Phoscon was not able to join devices after 20 min.

After give them the last try (RaspBee and ConBee I) I load the backup to a new rpi3b, same behavor. after ca. 20 min joining is impossible without powercycle or "sudo GCFFlasher_internal -r".

@manup: I don't know what the problem with the old FW/ HW/ Chipset is, but with the ConBee II it works, seems like something close the network (deny the join network) with the old ones... 
Is there an antenna orientation proposal for the new ConBee II ? It seems the range is a lot better.

closed 2019-12-15 by stale[bot]