dresden-elektronik/deconz-rest-plugin #1693: RPi 4 / Conbee 2

dhartmann84, 2019-07-20:
I have a ConBee 2 stick in a Raspberry Pi 4. Unfortunately I have problems because my Aqara temperature sender is not found. (WSDCGQ11LM)
 I have the latest Rasbian Buster installed.

For the installation I proceeded as follows:
1. sudo apt update && sudo apt -y upgrade
2. wget http://www.dresden-elektronik.de/rpi/deconz-latest.deb
3. sudo dpkg -i deconz-latest.deb
4. sudo apt install -f
5. sudo systemctl enable deconz

The browser can open the Phoscon app as described, but if I want to add a new sensor, nothing will happen. On the page stands for firmware "not connected" and version "2.05.66 / 16.6.2019".

Is there a bug in the software or am I doing something wrong?
Thank you very much for your help. 

Greetings from Switzerland.

Smanar, 2019-07-20:
Hello, you have manipulation to try for your situation here https://www.phoscon.de/en/support#conbee2-connection-issues

pedro-araujo, 2019-07-20:
I'm also having the same problem, I just bought a RaspberryPi 4 (I'm using Archlinux ARM) and I can't connect Phoscon to the Conbee 2, the status in the "Gateway" section simply says "not connected".

I'm pretty sure I have everything properly configured software wise because it was working on my old Raspberry Pi 3B+. My upgrade from the 3B+ to the Raspberry Pi 4 consisted in simply plugging the SD card from the old Pi to the new Pi without any software upgrades or changes in configuration at all.

I can see that the ConBee stick is properly detected by the OS as dmesg shows:

`$ dmesg
[22579.199047] usb 1-1.4: new full-speed USB device number 21 using xhci_hcd
[22579.337648] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[22579.337687] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[22579.337715] usb 1-1.4: Product: ConBee II
[22579.337735] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[22579.337757] usb 1-1.4: SerialNumber: DExxxxxxx
[22579.348420] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
`

And I am part of the same user groups as /dev/ttyACM0 and /dev/ttyAMA0 (in Archlinux it is uucp instead of dialout but it worked fine in the RaspberryPi 3B+).

This is the output log from deCONZ when starting up:

`22:50:08:075 CTRL. 3.29.022:50:08:173 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)                                   
22:50:08:173 COM: /dev/ttyAMA0 :  (0x0000/0x0000)      
22:50:08:173 dev /dev/ttyAMA0 
(...)
20:50:08:353 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin                                               
20:50:08:385 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)                                                            
20:50:08:385 COM: /dev/ttyAMA0 :  (0x0000/0x0000)                                                                     
20:50:08:385 dev /dev/ttyAMA0                                                                                                                                                                                                                
20:50:08:386 auto connect com /dev/ttyACM0                                                                            
This plugin does not support propagateSizeHints()                                                                                                                                                                                            
This plugin does not support propagateSizeHints()                                                                                                                                                                                            
20:50:08:407 Serial com connected                                                                                                                                                                                                            
20:50:08:413 discard sensor state push for 1: state/dark (already pushed)                                             
20:50:08:414 could not delete /var/tmp/deconz-firmware-version                                                        
20:50:08:416 Master: read param with arg 0x18                                                                         
20:50:08:416 Master: read param with arg 0x13
20:50:08:416 Master: read param with arg 0x13
This plugin does not support propagateSizeHints()
20:50:08:426 Device firmware version 0x26490700
20:50:08:427 discard sensor state push for 1: state/status (already pushed)
20:50:08:436 unlocked max nodes: 200
20:50:08:436 has feature: max nodes (0x11), status: 0x00
20:50:08:441 discard sensor state push for 1: state/lastupdated (already pushed)
20:50:08:441 has feature: std security (0x01), status: 0x04 
20:50:08:442 has feature: link security (0x02), status: 0x00
20:50:08:446 has feature: high security (0x03), status: 0x04
20:50:08:450 has feature: ethernet (0x04), status: 0x04
20:50:08:454 has feature: mass storage device (MSD) (0x05), status: 0x04
20:50:08:458 has feature: device firmware upgrade (DFU) (0x06), status: 0x04
20:50:08:462 has feature: ZigBee Light Link (ZLL) (0x07), status: 0x04
(...)
20:50:13:129 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)                                                       
20:50:13:129 COM: /dev/ttyAMA0 :  (0x0000/0x0000)                                                                     
20:50:13:129 dev /dev/ttyAMA0                                                                                         
20:50:13:129 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF
20:50:13:129 GW firmware version: 0x26490700 
20:50:13:129 GW firmware version is up to date: 0x26490700
(...)`

Is there anything useful here that might help to debug the problem? It is kind of annoying because it feels like deCONZ is capable of seeing the USB stick.

dhartmann84, 2019-07-23:
> Hello, you have somes manipulations to try for your situation here https://www.phoscon.de/en/support#conbee2-connection-issues

I followed all instructions but unfortunately without success.

Does anyone else have an idea?
I am slowly turning crazy.

Smanar, 2019-07-23:
> I'm pretty sure I have everything properly configured software wise because it was working on my old Raspberry Pi 3B+. My upgrade from the 3B+ to the Raspberry Pi 4 consisted in simply plugging the [...]

It don't works like that, if you take a sd from a pi N to a PI N+1, you NEED update it.

- sudo apt-get update
- sudo apt-get upgrade

and

- sudo apt-get install rpi-update
- sudo rpi-update

pedro-araujo, 2019-07-23:
I'm not using Debian, I'm using Arch. I'm fairly confident that I know what I'm doing OS wise.

In anyway, I suspected of USB 3.0 (I had heard that USB 3 speed could cause interferences with USB 2 devices) and I did some trials.

I had an external USB 3 HDD connected to the Raspberry Pi and I decided to disconnect it and I connected Conbee to one of the USB 2.0 ports (the black ones) and surprise surprise: it worked.

Maybe people can try the same? Disconnect any USB 3.0 devices and try Conbee in the USB 2.0 port?

Smanar, 2019-07-23:
Oups, sorry, bad reading.
And yes, you are right : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1598#issuecomment-508911286

dhartmann84, 2019-07-24:
Sometimes the solution is so close!  Thank you very much for the great help.  I connected the Conbee and the USB stick to the two USB 2.0 ports and now everything works!  It may not be anything to the USB 3.0 ports ... very special error.

closed 2019-07-24 by dhartmann84