dresden-elektronik/deconz-rest-plugin #1694: all lights are grayed out after server reinstall

DanielHD80, 2019-07-20:
Hello, I have a little disaster!

I've been hanging my entire apartment at Conbee II for weeks. So IKEA Tradfri bulbs and Zigbee sensors from Xiaomi (Xiaomi Smart Button). These are all my lighting in the apartment.

Yesterday I installed my new server (before and after Proxmox, VM was taken over by Backup / Restore!). On the new server, the USB device in the VM deleted and re-selected (was detected immediately). In the Phoscon app, the stick is also detected (firmware is displayed, which disappears when the stick is removed). Strangely, ALL my lights are grayed out! How can I get these back? Do I have to relearn the entire apartment? This can not be? Also a restart of the old server brought the same problem! Stick is detected, lamps are gray! Who can help me? The ConBee II is already awesome, but it can not be that you can not access the devices after a server restart?

Smanar, 2019-07-20:
You have done a Backup/Restore so you haven't changed your configuration files ?
Have you make too a backup of your previous deconz install, in phoscon ?
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1587

DanielHD80, 2019-07-20:
I made a backup on the old server and restored on the new one. no changes...
a backup in phoscon wasn't made, but the settings are there.... it's "only" without function
I found the recovery-tool here on github, tested it and shows all same config (all lines are similar)
I have power cycled many bulds and waited over 1h... no chance...

Smanar, 2019-07-20:
Try a previous "Zigbee configuration" even it tey look same.
Try to re-pair at least one bulb to check if you have not a bug in the new installation.

DanielHD80, 2019-07-20:
I made some reboots... Now the firmware is not connected... :(

Smanar, 2019-07-20:
Depend on machine, but on some of them (like raspberry, you to unplug/replug the device to reboot it, reboot the computer is not enought because the usb port is still powered)

DanielHD80, 2019-07-20:
It's a VM with Proxmox on an Intel NUC... Tested with replus, not connected... lsusb shows die conbee

Smanar, 2019-07-20:
Die conbee ? She is not listed ?

DanielHD80, 2019-07-20:
I don't think that it died... It was connected an hour ago... :(

closed 2019-11-24 by stale[bot]