dresden-elektronik/deconz-rest-plugin #1695: Network died overnight

bigbellyburger, 2019-07-20:
The network map still shows all devices, but nothing is connected and nothing will connect. I am using the ConBee II usb stick on a linux machine, and it's connected using an usb extension cable.

At first the log was filled with `error APSDE-DATA.confirm: 0xD0 on task`. I've since then restarted deconz a few times, rebooted the host system, and tried replugging various hardware devices. This error no longer shows up, but still nothing is connected.

I've also attempted a restore of the deconz data directory from two days ago when everything was working, this changed nothing.

Oddly enough some aqara sensors sometimes show as connected, but no events are received. I also just managed to restore connection to an ikea power socket by permitting new devices to join and factory resetting it.

My network consists of:

* Aqara sensors (some show as connected, but don't actually work)

* Philips hue bulbs (won't connect)

* Innr power sockets (won't connect)

* Ikea power socket (won't connect)

* Ikea repeater (won't connect)

I have no idea why this happened as it broke overnight while no software or hardware changes occurred.

Smanar, 2019-07-20:
Have you tried to unplug/replug the gateway ?
Have you an usb cordon to make tries with extension ?
On phoscon you have the same display https://www.phoscon.de/support/img/faq-conbee2-not-connected-en.png

bigbellyburger, 2019-07-20:
> Have you tried to unplug/replug the gateway ?

Several times, both with and without usb extension cable. Seemed to not make any difference. I've also rebooted the host machine.

> Have you an usb cordon to make tries with extension ?

Sorry, what do you mean? I'm currently using a regular USB extension cable.

>  On phoscon you have the firmware version displayed ?

[image]

And from the log: ```17:36:27:530 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF
17:36:27:530 GW firmware version: 0x264a0700
17:36:27:530 GW firmware version is up to date: 0x264a0700
```

Smanar, 2019-07-20:
>Sorry, what do you mean? I'm currently using a regular USB extension cable.
Was the test which I wanted to talk about.

The  ikea power socket is working ? The one you have re-added after ?

bigbellyburger, 2019-07-20:
> The ikea power socket is working ? The one you have re-added after ?

Yes, it is working after I've re-added it (search for new light in phoscon, press the button on the ikea switch for 5 seconds).

I have another identical ikea switch which I haven't reset, and that one isn't working.

Smanar, 2019-07-20:
> Interestingly it seemed to remember its name within deconz.

Yep, it's something normal, you can just re-pair all devices, without deleting them in phoscon to solve your problem (from my memory it's not perfect for somes sensors) , but I can understand it's not a "real" solution.

bigbellyburger, 2019-07-20:
What does `error APSDE-DATA.confirm: 0xD0 on task` mean?

closed 2019-11-24 by stale[bot]