dresden-elektronik/deconz-rest-plugin #1697: Names with single quotes cause problems

titanous, 2019-07-20:
Creating a group or renaming a device to a name with an apostrophe (aka single quote) (for example `Jonathan's Light`) will appear to save correctly, but when deCONZ is restarted the configuration will either revert to before the rename or the group/device will not be in the database. Using a modifier letter apostrophe (`ʼ`) instead works.

I expect this is because `'` is a special character in SQLite and there is incorrect escaping/parameter binding.

This is the cause of #1287.

ebaauw, 2019-07-21:
I can confirm:
```
$ ph put /lights/6 "{\"name\": \"Erik's test switch\"}'
```
results in a SQL error:
```
Jul 21 12:26:02 pi1 deCONZ[1121]: 12:26:00:384 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('6', 'normal', '00:0d:6f:00:0f:d8:56:6c-01', 'Erik's test switch', '65520', '1', 'ICZB-IW11SW', 'icasa', '2.4.1_r33', '{"attr/manufacturername":"icasa","attr/modelid":"ICZB-IW11SW","attr/name":"Erik's test switch","attr/swversion":"2.4.1_r33","attr/type":"On/Off light","attr/uniqueid":"00:0d:6f:00:0f:d8:56:6c-01","state/alert":null,"state/on":false,"state/reachable":true}')
Jul 21 12:26:02 pi1 deCONZ[1121]: 12:26:00:384 DB sqlite3_exec failed: REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('6', 'normal', '00:0d:6f:00:0f:d8:56:6c-01', 'Erik's test switch', '65520', '1', 'ICZB-IW11SW', 'icasa', '2.4.1_r33', '{"attr/manufacturername":"icasa","attr/modelid":"ICZB-IW11SW","attr/name":"Erik's test switch","attr/swversion":"2.4.1_r33","attr/type":"On/Off light","attr/uniqueid":"00:0d:6f:00:0f:d8:56:6c-01","state/alert":null,"state/on":false,"state/reachable":true}'), error: near "s": syntax error
```
I did a quick code scan, but there don't seem to be any WHERE clauses on user editable attributes, so I don't think we need to worry about SQL injection.

After patching the database, updating the name to `Erik's test switch`, it loads correctly and displays correctly in the GUI and in the REST API.

holdestmade, 2019-08-07:
Same here, have had this problem for months and not realised it was because of the single quote. I have simply removed the single quote and all works fine now - Simon's Switch is now Simons Switch

bfayers, 2019-08-12:
Also happens to me, been trying to work out for the last week why only 2 of my lights (and remotes) would revert their names and now I know why, thanks!

titanous, 2019-12-10:
I don't think this issue has been fixed. Do not close.

titanous, 2020-04-09:
I don't know if this issue has been fixed.

inthevidual, 2020-11-12:
I can confirm that this is till an issue with the latest versions of deConz/conbee II firmware as of today.

Mimiix, 2020-11-13:
Re-opening.

deviantintegral, 2021-02-07:
This is quite tricky for end users to figure out, given that the restart may be some time in the future after changing the name. If this is tricky or low priority to solve, could we at least get a warning in description text when renaming a device or group in the Phoscon UI, linking to this issue?

titanous, 2022-01-17:
Fixed in #5583

closed 2022-01-17 by titanous