dresden-elektronik/deconz-rest-plugin #1698: Xiaomi aqara wall switch 2 gang - no action on group light

sr-1989, 2019-07-20:
Hi !
I'm new on Github, I'm new on Deconz, i'm discovering this new universe with hassio :) (i'm a noob! lol)
first impressions are very very good!
i'm using some xiaomi modules and wall switch, i bought also ikea lights and i would like to have only one gateway, that why i'm moving my install on hassio with deconz :) .
But i've met a problem that i can't solve, and i passed all my day to try to find a solution on internet but nothing found...

i got the following problem : 
i appaired correctly (i think...) [my xiaomi 2-gang wall switch (with neutral) ](https://aws1.discourse-cdn.com/smartthings/original/3X/6/d/6d79e399a68a9d05b89f9585c42785e6145152a8.jpg) 

on phoscon, 2 lights are created, for the first light all is ok no problem, but for the second impossible to switch it with its group button, but when i click directly on the light inside the group the action is ok...
i'm embarrassed because i can't add other switch to toggle it because the action is made on the group and not on the light inside directly...

i can access to the deconz app with vnc, but i don't know if the problem is on this part...

Hope you could help me!
(i join a pic to try to show the problem)
[image]

Thanks in advance ! 

regards,

Sr

TroLoos, 2019-07-21:
I have exactly the same issue...

For my 2-way switch - only Left button reacts on group action, the other one doesn't. I even tried to separate right button to different group, no change at all.

I remember this worked quite well couple of versions ago.

sr-1989, 2019-07-21:
> I have exactly the same issue...
> 
> For my 2-way switch - only Left button reacts on group action, the other one doesn't. I even tried to separate right button to different group, no change at all.
> 
> I remember this worked quite well couple of versions ago.

yes exactly, problem on the right button for me too...
hope someone have a solution ! :s

sr-1989, 2019-08-08:
hi, no hints about it ?
thanks!

Chapayyy, 2019-09-01:
hi! I have this problem too. What to do?

TroLoos, 2019-09-02:
Yeah... it seems like development is focused on delivering compatibility to new devices (after looking at commits). I have several bugs with Xiaomi switches, this is just one of them and one I can even live with.

What I can't live with is a case where all my single button switches just turn off after random time. It looks like the support for them is broken. They constantly report power usage even when off, etc. What happened to dev team that they don't notice it at all?

prebrov, 2019-11-02:
Have the same issue with a single gang no neutral Aqara switch.

Would appreciate any guidance or ideas.

ebaauw, 2019-11-03:
> Would appreciate any guidance or ideas.

Don't use Xiaomi wired switches.  These issues are caused by the Xiaomi firmware, not by deCONZ.  They simply don't react to group commands, or only on the first endpoint.

> What I can't live with is a case where all my single button switches just turn off after random time.

I've seen this on multiple `lumi.ctrl_ln2.aq1` switches.  As far as I've been able to determine, this is caused by faulty hardware.  See https://github.com/ebaauw/ceiling-fan.

TroLoos, 2019-11-04:
> > Would appreciate any guidance or ideas.
> 
> Don't use Xiaomi wired switches. These issues are caused by the Xiaomi firmware, not by deCONZ. They simply don't react to group commands, or only on the first endpoint.

Hmmm, nice advice... especially when you have 30+ installed :-) To be honest, I'm not so sure about this as this used to work quite well sometime ago with previous Deconz version and Conbee stick firmware.

> > What I can't live with is a case where all my single button switches just turn off after random time.
> 
> I've seen this on multiple `lumi.ctrl_ln2.aq1` switches. As far as I've been able to determine, this is caused by faulty hardware. See https://github.com/ebaauw/ceiling-fan.

That's absolutely not true. These switches work very well with original Xiaomi Gateway, as well as with Zigbee2MQTT - no problems whatsoever. I'm pretty sure that it's not a hardware issue.

To be honest, I've been forced to move out all my Zigbee devices (approx. 50 of them) from Deconz to original gateway. Deconz seems very unstable with LOTs of bugs that are not taken care of.

Examples:
- Xiaomi single switches turn off after couple of seconds, random time, but happens always once turned on
- some Xiaomi wired double switches started to show up in Phoscon app as Temperature sensors, and report low battery via REST plugin
- devices go offline even though coverage of Zigbee is guaranteed @ home where each room has at least 1 wired Zigbee router, it also happened to those routers not only battery devices

ebaauw, 2019-11-04:
> > They simply don't react to group commands, or only on the first endpoint.

> To be honest, I'm not so sure about this as this used to work quite well sometime ago with previous Deconz version and Conbee stick firmware.

There are many different Xiaomi switches out there, all looking alike, but with different hardware and/or firmware.  You never know which one you'll get when ordering them on AliExpress.  I know for a fact my `lumi.ctrl_neutral` switches (oddly without neutral wire) don't react to group commands.

If your switches used to react to group commands before, but don't anymore, most likely they've dropped off the group.  Could be because of a device reset, see below.  You can check in the _Cluster Info_ panel for the _Groups_ cluster in the deCONZ GUI, if they're still a member of the group.

> > I've seen this on multiple `lumi.ctrl_ln2.aq1` switches. As far as I've been able to determine, this is caused by faulty hardware.

> That's absolutely not true.

I think, in my case, it is.  They also turn off when not joined to any network.  I don't have a Xiaomi gateway to check, though.

> These switches work very well with original Xiaomi Gateway, as well as with Zigbee2MQTT.

What hardware/ZigBee gateway do you use with ZigBee2MQTT?

> Deconz seems very unstable with LOTs of bugs that are not taken care of.

I'm afraid that comes with the territory of trying to support a multi-manufacturer ZigBee network. Especially with devices that don't adhere strictly to the ZigBee standard.  Their own gateway/hub/bridge will always be more reliable.

> Xiaomi single switches turn off after couple of seconds, random time, but happens always once turned on

The `lumi.ctrl_neutral` switches query the _Time_ cluster on the coordinator every minute or so, and reset when they don't get an answer, see #757.  They use relays that need continuous power on the control signal to remain on.  A device reset interrupts this signal and causes the relay to switch off.

Could you sniff the ZigBee traffic between the switch and the Xiaomi gateway to see if your switches use a similar heartbeat.  Alternatively, could you turn on excessive logging on deCONZ, and check if it receives regular device announcement messages from the switch.  This would confirm a device reset.

Just for completeness: did you check the load on the switches?  I think my `lumi.ctrl_ln2.aq1` switches blew on the inductive load of the AC motors in my ceiling fans.  The `lumi.ctrl_neutral` don't seem to mind this at all (knock on wood).  Does the switch also turn off randomly, when nothing is connected?

> some Xiaomi wired double switches started to show up in Phoscon app as Temperature sensors, and report low battery via REST plugin

Typically, Xiaomi devices report their internal device temperature through the Xiaomi special attribute report, but that shouldn't trigger the creation of a `/sensors` resource.  Could you list the corresponding resources?  What version of deCONZ are you on?  Is that the same version you used to pair these switches (or tried to pair new devices, after which the sensors appeared).  

> devices go offline even though coverage of Zigbee is guaranteed @ home where each room has at least 1 wired Zigbee router, it also happened to those routers not only battery devices

What makes you think the device goes offline?  I'm still having occasional issues where deCONZ loses the route to my `lumi.curtain`, see #849.  The device is online alright, but deCONZ cannot reach it over unicast.

For completeness: could also be caused by radio interference.  Do you use a USB extension cable for the ConBee?  It is connected to a USB2 port?  No USB3 devices nearby?  Wifi and bluetooth disabled?

closed 2020-03-10 by stale[bot]