dresden-elektronik/deconz-rest-plugin #1702: Device support request for Zen Thermostat

arakasi55, 2019-07-23:
Hi, I have a Zen Thermostat that I'd like to have supported by DeConz. 
Here are the node and cluster info screenshots.
Thanks.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

MarhyCZ, 2019-10-20:
Hi, I was just looking for a thermostat to use in HomeAssistant. Either through Wifi or ZigBee. This Zen one has a really nice design. Are you planning on adding this to REST API? 

Have a nice day :)

ggg22, 2020-01-07:
Hi, I wonder if the Zen thermostat is fully supported. Google searches suggest that is not the case since I haven't seen anybody posting a successful implementation.
While I am able to join the network with the Zen thermostat, the device is not visible in the Phoscon app. From there I plan to use it in Homeseer but at the moment this is not possible.
Can anybody comment, please?

[image]

closed 2020-05-13 by stale[bot]

crixyd, 2020-07-30:
For anyone who is reviewing this ticket to confirm support, I can confirm full ZEN thermostat compatibility for temperature reporting and control & mode reporting and control. It's rock solid, and vastly superior to the MQTT implementation 👍

ggg22, 2020-07-30:
Thanks very much for this update! Much appreciated!

 

Regards,

-George

ggg22, 2020-08-25:
> Reviewing your previous comment @ggg22 <https://github.com/ggg22>  , the ZEN doesn't appear in Phoscon UI for me either, however I think that's unnecessary as it is fully supported by the Deconz REST API so should work just fine with downstream integrations. For example I'm using Home assistant and the device is exposed with full compatibility out of the box now.
>
> Two FYIs
>
> *	Full device support was implemented in the most recent Deconz beta version 2.05.79
> *	There is a configuration issue in 2.05.79 beta (which I suspect will be fixed in stable) whereby temperature isn't reported from device to HA by default, however it's simply resolved by updating Thermostat Cluster attribute 0x001C 'System Mode' as follows (via VNC interface)
> -- Min Report Interval = 1
> -- Max Report Interval = 600

Thanks for the tips.

I still have to update to 2.05.79 and also figure out how to use VNC interface to modify the report intervals.

I noticed that it’s not updating temperature from the device.

The “set point” is updated TO the device but nothing is updated FROM the device.

If you change the “set point” temperature on the device itself this is not updated to Phoscon either.

 

Thanks,

-George

barryl93, 2022-10-29:
Does this device support switching mode ("Off" to "Heat" for example) through Zigbee? I have it set up through Homebridge-Hue and it communicates the temperature and set point in both directions, but if I switch from Heat to Off or vice versa on my phone, the mode doesn't switch on the thermostat. If I switch from Heat to Off on the thermostat, it WILL update correctly on my phone, though.

Does anyone have any experience with this device and this issue?

crixyd, 2022-11-21:
This used to work, however somewhere along the line they updated the integration and now to turn off the unit you just need to set the temp to the lowest possible. Still works but not as well as it used to.

barryl93, 2022-11-21:
Thanks for the confirmation. Yeah, now I just crank the temp way down to emulate "Off." Kinda frustrating, though, that I can't just say, "Hey, Siri, turn off the heat!" She'll turn it off on my phone, but it won't actually change anything on the thermostat. So  I just have to keep that in mind.

ericyew, 2024-04-30:
don't understand why the control doesn't work. Can someone have a look at this?