dresden-elektronik/deconz-rest-plugin #1703: HTTP ERROR 413 Phoscon Web App URLs too long

mercenaryballs, 2019-07-23:
I am having a weird issue. I am running deconz v2.05.66 in a docker on my unraid server and I cam getting an error when clicking links in the Phoscon Web App.

Clicking any of the links on the side bar trys to open a url like: 
> http://192.168.10.14:9090/pwa/settings-view.html?_v=6683c360fca0884412320c2573ce62c0c2f73ac7,6,2,18&gwid=00212EFFFF013EF7

This produces and a HTTP Error 413 in Chrome. If I shorten the url to 

> http://192.168.10.14:9090/pwa/settings-view.html

I can load the page, but the issue is reproduced as soon as I click another link. This happens for all links clicked on the page. The only issue I see in the deconz log is stating the websocket was disconnected.

[image]

mercenaryballs, 2019-07-24:
I am not sure why, but I have no issues when usin the web app from my laptop on Chrome as well. I thought a 413 error would be a server side issue. Now I am confused.

mercenaryballs, 2019-07-26:
I'm closing this because it magically went away. It must have been something on my end.

closed 2019-07-26 by mercenaryballs

KoljaHenckel, 2021-05-28:
I have been using Phoscon for about two years now on my homeassistant (RP4). Last week it stoppen working with this same 413 http error. All other Conbee software is still working, but the phoscon gui cannot load :(
I a quite confused what to do..?

sydragos, 2021-05-30:
Cleared cookies and it worked..

jamieshaw, 2021-06-28:
I'm hitting this myself when attempting to access remotely via Apache2 reverse proxy.

The same URL, internally, pointing directly to the host, has no such issue. And even with a clean/private session, the issue persists. 🤔

bvelte95, 2022-06-09:
I also suffer the same issue with deconz docker on Unraid. Did anybody find a solution yet?

AlexKalopsia, 2022-07-15:
Also experiencing this. Doing `Ctrl` + `F5` to clear cache didn't fix the issue. 

Changing browser, however, did.

I am getting `413` only on the phoscon webapp (not on the novnc)

Oscar438, 2022-07-30:
Clearing cookies worked for me:

In chrome when on the page:
`F12` -> `Application` -> Expand `Cookies` -> Right click the URL and click `clear`
[image]

IIIdefconIII, 2023-03-12:
issues still excist, anyone got a solution? tried several browsers + incognito. cleared cache and cookies. 22 tunnels working excapt deconz. local is running fine

Mimiix, 2023-03-12:
Locking this up. Phoscon issues are tracked on the forums .