dresden-elektronik/deconz-rest-plugin #1705: Question/Feature Request: Modify state of soft-swiched-off lights

githtz, 2019-07-25:
I want modify the brightness and the color temperature of lights that are soft-swiched-off but powered and able to receive commands. I know this is currently not possible by using the phoscon app, but is it possible at all (considering the API and the zigbee-devices). 

Use-case: 
I'd like to create some rules that automatically reduce the brightness of lights at nighttime and warm up color temperature, too. This works fine of all switched-on lights, but the others will still shine bright and cold when switched on.

ebaauw, 2019-07-25:
Technically, ZigBee doesn't support changing the light state when it's off (even though the light is still powered).  So you either need to change the state _before_ turning it off, or _after_ turning it on.  Depending on how quickly the light receives and handles the commands, that might look ugly.  The REST API plugin currently doesn't handle this well, see #1111. 

The surest way to achieve what you want is to turn the lights on by recalling a scene, using different scenes for daytime vs nighttime.  This requires a series of rules on the gateway and a means to trigger the rule, either a physical or a virtual switch (CLIPGenericFlag sensor).

githtz, 2019-07-25:
@ebaauw Thanks, I'll try that! Sad though.

closed 2019-08-02 by githtz