dresden-elektronik/deconz-rest-plugin #1706: Unable to update firmware of Conbee

skeptizzle, 2019-07-25:
Hi.

I'm having trouble updating my ConBee firmware. deCONZ is version 2.05.66 but still with firmware version 260B0500.

Running through Home Assistant add-on store on a NUC.
Tried to update using Phoscon GUI, but the version just rolls back to 260B0500.
Then i tried to remove my ZWave USB adapter as mentioned in the add-on docs but no dice.

App claims that the update was performed successfully, but when I check in deconz GUI, it still shows 260B0500.
Tried to update manually following the instructions on Windows 10, then Raspbian, then Windows 10 again.

Tried a earlier version of GCFFlasher (because it was all i could find) and that resulted in "error 828 expected dev0.serialNumber not empty", and GCFFlasher -l showed no serial or path.

Later i got a hold of version 3.05 (in this package https://www.dresden-elektronik.de/deconz/win/deCONZ_Setup_Win32_V2_05_66.exe), some improvement, but it's here I'm stuck. I keep getting the following on both Windows 7 and 10 (Windows XP didn't work at all):

C:\Users\...\deCONZ\bin>GCFFlasher -d COM4 -f ../firmware/deCONZ_Rpi_0x26330500.bin.GCF
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device COM4 (ConBee)
deCONZ firmware version 260B0500
reset FTDI 5&C1BD4D0&0&2 (dev: 0)
action: update firmware after 4 ms
could not sync with bootloader

Is it time to buy a ConBee II? I have burned most of the day today so any help og clarification is highly appreciated!

heine79, 2019-07-29:
Same issued here, tried USB 2 & USB 3 Hub. 

When tried to Update via phoscon deconz is showing to progress bar counting to 100% nur it's still in old Firmware. 

Via cmd line Error:
could not sync with bootloader

Any Chance to Upgrade?

BG,

manup, 2019-07-29:
Hi, please check this wiki page for manual update instructions:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

heine79, 2019-07-29:
Wasn't able to upgrade via Windows (app or manuell). What was working wor me now: 
In Hassio changed to official deconz add on. In this one i was able to update firmware.

skeptizzle, 2019-07-29:
Hi @manup

I've followed the guide both for Windows and Raspbian, same result.. 😒

manup, 2019-08-01:
Not sure what the issue is yet here are some commands which hopefully yield more information:

    GCFFlasher -d COM4 -x 3 -f ../firmware/deCONZ_Rpi_0x26330500.bin.GCF

The -x 3 parameter enables more debug logging.

Also interesting would be the output of:

    GCFFlasher -l

skeptizzle, 2019-08-02:
@manup let me try this out and get back. I'm juggling a lot of work right now so i dont know when. 
Thanks!

Wegatarus, 2019-08-07:
Hi, I have the same problem and try the -x3 parameter:
pi@WEGADIESSEN:~ $ sudo GCFFlasher_internal -d /dev/ttyUSB1 -x 3 -f deCONZ_Rpi_0x26330500.bin.GCF
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
using firmware file: deCONZ_Rpi_0x26330500.bin.GCF 
ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
ls dev: /dev/ttyUSB0 (0x0403/0x6001) sn: 00000000
ls dev: /dev/ttyUSB1 (0x0403/0x6015) sn: DM00ZEM0
ls dev: /dev/ttyUSB2 (0x0403/0x6001) sn: A9M9DV3R
ls dev: /dev/ttyUSB3 (0x0403/0x6001) sn: 433
search FTDI device: no.1
found 4 FTDI devices
Reboot device /dev/ttyUSB1 (ConBee)
serial port open device
deCONZ firmware version 261F0500
set watchdog ttl status: 0x04
close serial port
uart reset failed, try FTDI reset
action: reset device FTDI
failed to open FTDI device (0) status = 3

ftdi_sio
failed to open FTDI device (1) status = 3

ftdi_sio
failed to open FTDI device (2) status = 3

ftdi_sio
failed to open FTDI device (3) status = 3

ftdi_sio
reset FTDI DM00ZEM0 (dev: -1)
FTDI can't open device -1, status = 2

With GCFFlasher -l Iget
sudo GCFFlasher_internal -l                                          GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh                           Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee 
/dev/ttyUSB1     | 0x0403 | 0x6015  | DM00ZEM0   | ConBee
Thanks

manup, 2019-08-08:
Hmm not sure if it is related to the other FTDI devices.
Can you please try to update the ConBee while the other USB devices aren't connected?

macce76, 2019-08-08:
Same problem here,

macce76, 2019-08-08:
not working without connbee

macce76, 2019-08-08:
This is mine in a win10 machine:
C:\Users\AppData\Local\deCONZ\bin>GCFFlasher -d COM5 -x 3 -f ../firmware/deCONZ_Rpi_0x26330500.bin.GCF
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
using firmware file: ../firmware/deCONZ_Rpi_0x26330500.bin.GCF
ls dev: \\.\COM1 (0x0000/0x0000) sn:
ls dev: \\.\COM3 (0x8086/0x1E3D) sn:
ls dev: \\.\COM5 (0x0403/0x6015) sn: DO00KMQ8A
Reboot device COM5 (ConBee)
serial port open device
deCONZ firmware version 260B0500
set watchdog ttl status: 0x04
close serial port
uart reset failed, try FTDI reset
action: reset device FTDI
reset FTDI DO00KMQ8A (dev: 0)
action: update firmware after 2 ms
serial port open device
STARTING APP
 (395 ms)
STARTING APP
? (475 ms)
STARTING APP
? (2605 ms)
could not sync with bootloader
close serial port

macce76, 2019-08-08:
Suddenly it worked update in phoscon app!!!
So keep trying.
There are three downloads for win try the another one.

sdrmlm, 2019-08-08:
I have this problem when running deconz in an fully updated docker container on Ubuntu 18.04 x86. My current ConBee version is 260B0500. 

Following the instructions for manually upgrading in docker gives me the error: "828: Error: (assert) expected dev0.serialNumber not empty". 

`/usr/bin/GCFFlasher_internal -l` returns:

´´´
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyUSB0     | 0x0403 | 0x6015  |            | ConBee
´´´

Wegatarus, 2019-08-11:
hi, here the output without any other usb device:

 sudo GCFFlasher_internal -d /dev/ttyUSB0 -x 3 -f deCONZ_Rpi_0x26320500.bin.GCF
GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
using firmware file: deCONZ_Rpi_0x26320500.bin.GCF
ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn:
ls dev: /dev/ttyS0 (0x0000/0x0000) sn:
ls dev: /dev/ttyUSB0 (0x0403/0x6015) sn: DM00ZEM0
search FTDI device: no.0
found 1 FTDI devices
Reboot device /dev/ttyUSB0 (ConBee)
serial port open device
deCONZ firmware version 261F0500
set watchdog ttl status: 0x04
close serial port
uart reset failed, try FTDI reset
action: reset device FTDI
failed to open FTDI device (0) status = 3

ftdi_sio
reset FTDI DM00ZEM0 (dev: -1)
FTDI can't open device -1, sta

closed 2019-12-16 by stale[bot]