dresden-elektronik/deconz-rest-plugin #1707: Support for Heimann Smart remote Control HS1RC

accelle17, 2019-07-26:
support for heimann hs1rc remote [http://www.heimantech.com/product/95.html](url)

It's already requested on a stale closed issue #526 .

Here are the screenshots and verbose logs. I tried click all the buttons while the logs were set to verbose. I'm using hass io 0.95.4 on raspberry pi 3b with conbee2 stick

[deconz_hs1rc_logs.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3434565/deconz_hs1rc_logs.txt)
[image]
[image]
[image]
[image]
[image]
[image]
[image]

accelle17, 2019-07-29:
anything needed for this to be worked on? I will be glad to provide more troubleshooting / test

Pawnies, 2019-08-13:
Interested too !

accelle17, 2019-08-22:
any chance that this can be looked into?

Michelobb, 2019-12-12:
Hi !  Interested too.
Michel

beneffm, 2020-03-26:
Hi!
I also have bought 3 of the remotes to use them with my Conbee stick. Is there any chance to get them running, or how can I help to get them running :) ? 
Thanks a lot!

olix74, 2020-04-07:
Hi,
I would appreciate support for the Heiman Smart remote too.
Thank you!

ebaauw, 2020-04-07:
Start by adding the 0x0501 cluster (I think _IAS Ancillary Input_?) to `general.xml` following the ZCL specification.   Without that, deCONZ won't even log messages from this cluster.

accelle17, 2020-04-08:
@ebaauw thanks for the guide, is this something similar like adding <cluster id="0x0501" name="IAS Ancillary Control equipment">some items here</cluster> on general.xml?

can I just add it as custom xml on ZCLDB preference?

ebaauw, 2020-04-08:
Just edit `general.xml` and restart deconz.  Be sure to make a backup - if all clusters suddenly shown as unknown, you’ve made an error in the xml.

accelle17, 2020-04-09:
was able to add on general.xml and the 0x0501 cluster went from unknown to the defined name by adding the following line:

`<cluster id="0x0501" name="IAS Ancillary Control Equipment">
 <description>The IAS Ancillary Control Equipment cluster provides an interface to the functionality of any control device of the IAS system.</description>
<server>
</server>
<client>
</client>`

However, I couldn't see any logs about this cluster. I also checked the GUI and the node doesn't light when pressing the buttons.
[image]

Will research more about it.

ebaauw, 2020-04-09:
Doesn’t the cluster define any commands and/or attributes?  You might have to enroll the device before it sends anything.  I would guess this cluster would support that simehow, just as the _IAS Zone_ cluster.

accelle17, 2020-04-09:
isn't it already partially enrolled if I see it on vnc view? I was hoping I could see what it is sending in the logs so I could add commands/attr on the cluster definition.
It also looks like zigbee2mqtt dropped the support for this device due to an issue.

ebaauw, 2020-04-09:
Enrollment is the _IAS_ equivalent of setting up the bindings and attribute reporting.  This is normally handled by the REST API plugin, once support fir the device has been added.  It’s the step after the device joining the network (handled by the coordinator firmware) and deCONZ (the core programme) reading the descriptors to display the node in the GUI.

SwoopX, 2020-04-19:
#2697

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

Michelobb, 2020-07-20:
Hello,

Is the Heiman hs1rc-M now supported ?

Thank you,

Michel

Mimiix, 2020-07-20:
@Michelobb According to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2697  it is.

Michelobb, 2020-07-20:
Oh yeah !  I just received the deconz mail, specifying it is !  I will just have to wait for the update in Home Assistant :)

Thank you for your reply.

Michel

Mimiix, 2020-07-20:
@Michelobb As far as i know, it should work in HA. Latest version there is .78.

Michelobb, 2020-07-20:
@Mimiix 

I tried but the device can't be initialized (nor seen) at the moment.  The deCONZ latest mail about "Ghost Town" v02_05_79 announces  : 

**"Added initial support for Heiman remote (RC-EM) (#3004)"**

So I figured we'd have to wait for the .79.

But maybe I missed something, of course.

Mimiix, 2020-07-20:
@Michelobb That's a different one than the one in this issue. This issue  is about HS1RC

Michelobb, 2020-07-20:
@Mimiix 

Oh thanks Dennis.  So I need to check my HA out again...

joschaschultze, 2020-07-27:
I installed Beta79 and synced my Heiman Remote with Phoscon (as Sensor). I then was able to add the new device to domoticz. Now I have the following problem: When I press a key on the remote there are 4 events triggered:

2020-07-27 14:20:14 | B3
2020-07-27 14:20:09 |  
2020-07-27 14:20:03 | B3 
2020-07-27 14:19:59 |   
2020-07-27 14:19:55 | B3 
2020-07-27 14:19:49 |   
2020-07-27 14:19:47 | B3 

This is one press of button B3. If I press B3 and after that I immediately  press B2 than B2 gets triggered 4 times and B3 is only triggered once. 

I can "detect" these false presses with a timeout but it's a strange behaviour anyways.

olix74, 2020-08-13:
Hi, I also have .79 installed and I'm seeing the same issue as joschaschultze - thus I do receive 4 events on a single button press. Is there a way to change this?
Thank you very much!

rweise, 2020-09-06:
Why is this closed?
I can add the device and see them in the deconz-gui but in Phoscon there is nothing and the same in IOBroker :-(

joschaschultze, 2020-09-06:
@rweise  It is working with deconz v2_05_79 but you'll get 4 presses everytime you press a button.

rweise, 2020-09-06:
I run under v2_05_80 and why it should not work there?

Mimiix, 2020-09-06:
I'll ask a dev to look at this one and come back to me if it is implemented correctly.

Mimiix, 2020-09-06:
@rweise Which one do you have. The HS1RC-M or the HS1RC?

You are double posting...

joschaschultze, 2020-09-06:
@rweise: This is not a forum. But I can tell you that it's working with iobroker as "sensor". You have to pair it in phoscon first but it will not show up in phoscon. You can only use it in 3rd party applications like iobroker or domoticz through deconz.

SwoopX, 2020-09-06:
The reason why it's not working is most likely that you've not checked my referenced issue. You need to expose IAS ACE cluster on the coordinator, at least for the older device, or it leaves the network again after few seconds.

rweise, 2020-09-06:
I must say sorry, once for the double posting and second because i'm wrong.
I had ordered a HS1RC but if i look in the Basics it shows that it ist not a HS1RC
it shows me 
Manufacturer Name _TYZB01_qm6djpta
Modell Identifier TS0215

I take ashes over my head , sorry again ...