dresden-elektronik/deconz-rest-plugin #1711: HEIMAN HS1SA-E not working

filipvh, 2019-07-27:
I bought a HEIMAN HS1SA-E, and tried to add  it through the web app (the PWA), as a sensor. It detects the device, but does not show it listed. When I take a look in the Home Assistant integration, the device shows up, but it seems to not work :(
Is this a new "model" that needs to be whitelisted? Can I do this myself to get things running? How can I help to get it in the next version of deconz?

Smanar, 2019-07-27:
Perhaps like this one > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1345/commits/d8ead22ddb1ee5c999283783eb842ab6a42fe805 if it's just a new model.

filipvh, 2019-07-27:
That talks about a gas sensor. This is a smoke sensor.
[image]
And while home assistant picks it up, no signal is given when the alarm goes off :(

Smanar, 2019-07-27:
Was just an exemple, take a look on the static const SupportedDevice supportedDevices[] variable, you have lot of devices, perhaps yours is not inside.

filipvh, 2019-07-27:
They (3) seem to timeout and send no data :(
[image]

ebaauw, 2019-07-27:
You need to trigger the sensor, so it turns its radio on, before/while reading the cluster attributes.  It usually takes some trial & error.

filipvh, 2019-07-27:
[image]
funny thing is, my third sensor suddenly started to work properly. I don't know how I did that. It now shows detections in Home Assistant when it detects smoke (test button). It's not in the phoscon web app, but thats ok. They all seem to be willing to connect now, but only that 3rd one seems to "detect" :/

ebaauw, 2019-07-27:
I think these are IAS devices?  You might need to enroll them by writing the coordinator mac address to the _IAS_CIE_address_ attribute (0x0010) of the _IAS Zone_ cluster (0x0500).

filipvh, 2019-07-27:
@ebaauw That did the trick! 👍  thx! can you tell my why and what this is?

ebaauw, 2019-07-27:
The quickest way to kill a miracle is asking how it came to be ;-)

I’m not quite sure, actually, but that’s how it’s usually done.  There is an _Enroll_ command in that cluster, but I’ve yet to see a device that supports it.  The REST API plugin should do the enrollment upon pairing, but that kind of hit & miss.  The good news: once you’ve done it manually, it’s very stable.

filipvh, 2019-07-28:
[image]
so, they are returned by the REST interface, but the phoscon app does not know what to do with them... it seems

faxioman, 2019-11-19:
Sorry @ebaauw, but what does it mean "write the coordinator mac address to the IAS_CIE_address attribute (0x0010) of the IAS Zone cluster (0x0500)"?
I'm using deconz on a raspberry and I haven't the qt gui. Thanks!!!

closed 2020-03-26 by stale[bot]