dresden-elektronik/deconz-rest-plugin #1712: Cannot update RaspBee firmware

jaxer, 2019-07-27:
I followed this: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian
(after flashing prebuilt SD image on pi3).

Here is output:
```pi@phoscon:~ $
pi@phoscon:~ $ sudo GCFFlasher_internal -f deCONZ_Rpi_0x26330500.bin.GCF
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
close serial port
action: update firmware after 2 ms

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
 (103 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
S (506 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
ST (506 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STA (506 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STAR (506 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
START (507 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTI (507 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTIN (507 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING (507 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING  (508 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING A (508 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING AP (508 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP (508 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
 (509 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (591 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (591 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (592 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (592 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (592 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (595 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (595 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (595 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (596 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (596 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (597 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (2692 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (2692 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (2692 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (2692 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (2692 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (2693 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (2693 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (2693 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (2693 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (2694 ms)

RaspBee Bootloader premium
Vers. 1.02
build 2013/08/01
STARTING APP
� (2694 ms)
could not sync with bootloader```

All deconz apps are shut down.

Can you recommend what else i can try?

manup, 2019-07-28:
Can you please check step 1) from the installation to be sure that the serial interface is configured correctly:

https://phoscon.de/en/raspbee/install#raspbian

Some questions:
* Are there any other devices connected to the Raspberry Pi?
* Do you use a power-supply of at least 2.5A?

jaxer, 2019-07-28:
Thank you for quick reply. 

I am using freshly flushed SD image from https://phoscon.de/en/raspbee/sdcard (i guess those are already with uart enabled). To confirm i tried `raspi-config` according to manual. That made no difference.

I am using official rpi power supply (2.5A).
To confirm i tried with couple others with exactly same results.

Does it look like bootloader is not entering firmware upload mode of some sort? I should have JTAG programmers - should i try them? is there a manual somewhere to get me started?

manup, 2019-07-29:
Hmm ok, thanks for verifying. Actually the bootloader is active but somehow doesn't work through the update. Can you please retry with the latest GCFFlasher included in latest deCONZ version 2.05.66.

Install instructions are here: http://phoscon.de/de/raspbee/install#raspbian

It includes GCFFlasher 3.5 as well as firmware version 0x26330500

jaxer, 2019-07-29:
i cannot find SD image for version 2.05.66 (max 2.05.64).
so i installed deconz manually. 

but where can i find GCFFlasher 3.5? it is not installed in /usr/bin it seems.. ?

jaxer, 2019-07-29:
or maybe you meant GCFFlasher 3.05?

anyways i saw that Docker images are updated more often so i installed Docker and run flash script from there. Here is output:

```
pi@phoscon:~ $ sudo docker run -it --rm --entrypoint "/firmware-update.sh" --privileged --cap-add=ALL -v /dev:/dev -v /lib/modules:/lib/modules -v /sys:/sys marthoc/deconz
Unable to find image 'marthoc/deconz:latest' locally
latest: Pulling from marthoc/deconz
2eead4197fac: Pull complete
8153c3bca43e: Pull complete
14ea784a6e70: Pull complete
d72fc803ed1c: Pull complete
eaa717024ffe: Pull complete
9a030514b0a1: Pull complete
503daa6aaad2: Pull complete
47f594c92ae1: Pull complete
068c8d1e96ba: Pull complete
c99abb6057dc: Pull complete
428fe717b0b0: Pull complete
79a5f8b3baf3: Pull complete
1de46aae08c0: Pull complete
8bd964d3c64a: Pull complete
ea99fa54a397: Pull complete
4344667c8f17: Pull complete
Digest: sha256:4b6c21bb931df309d0c843579e24d48955cc58722daad75f12ec6bc86bcfada7
Status: Downloaded newer image for marthoc/deconz:latest
-------------------------------------------------------------------

             marthoc/deconz Firmware Flashing Script

                       Version: 0.5

-------------------------------------------------------------------

Listing attached devices...

GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee

Enter the full device path, or press Enter now to exit.

Device Path : /dev/ttyS0

-------------------------------------------------------------------

Firmware available for flashing:

deCONZ_ConBeeII_0x26480700.bin.GCF
deCONZ_ConBeeII_0x26490700.bin.GCF
deCONZ_Rpi_0x261f0500.bin.GCF
deCONZ_Rpi_0x26320500.bin.GCF
deCONZ_Rpi_0x26330500.bin.GCF

Enter the firmware file name from above, including extension,
or press Enter now to exit.

File Name : deCONZ_Rpi_0x26330500.bin.GCF

-------------------------------------------------------------------

Device: /dev/ttyS0

Firmware File: deCONZ_Rpi_0x26330500.bin.GCF

Are the above device and firmware values correct?
Enter Y to proceed, any other entry to exit: Y

Flashing...

GCFFlasher V3_05 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyS0 (RaspBee)
action: reset device RaspBee
action: update firmware after 2 ms
��)� (118 ms)
��)�S (505 ms)
��)�ST (506 ms)
��)�STA (506 ms)
��)�STAR (506 ms)
��)�START (507 ms)
��)�STARTI (507 ms)
��)�STARTIN (507 ms)
��)�STARTING (507 ms)
��)�STARTING  (507 ms)
��)�STARTING A (508 ms)
��)�STARTING AP (508 ms)
��)�STARTING APP (508 ms)
��)�STARTING APP
 (508 ms)
��)�STARTING APP
� (591 ms)
��)�STARTING APP
� (591 ms)
��)�STARTING APP
� (592 ms)
��)�STARTING APP
� (592 ms)
��)�STARTING APP
� (592 ms)
��)�STARTING APP
� (595 ms)
��)�STARTING APP
� (595 ms)
��)�STARTING APP
� (596 ms)
��)�STARTING APP
� (596 ms)
��)�STARTING APP
� (596 ms)
��)�STARTING APP
� (597 ms)
��)�STARTING APP
� (2679 ms)
��)�STARTING APP
� (2679 ms)
��)�STARTING APP
� (2679 ms)
��)�STARTING APP
� (2680 ms)
��)�STARTING APP
� (2680 ms)
��)�STARTING APP
� (2680 ms)
��)�STARTING APP
� (2681 ms)
��)�STARTING APP
� (2681 ms)
��)�STARTING APP
� (2681 ms)
��)�STARTING APP
� (2681 ms)
��)�STARTING APP
� (2682 ms)
could not sync with bootloader

Flashing Error! Please re-run this script...```

manup, 2019-08-01:
>but where can i find GCFFlasher 3.5? it is not installed in /usr/bin it seems.. ?

The binary is called `GCFFlasher_internal` on Linux please refer to
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian

For the invocation of the command. Note that the Docker container must be stopped before running the command on the host system.

closed 2019-12-07 by stale[bot]