dresden-elektronik/deconz-rest-plugin #1714: [Bug] All Group not adding devices

Ltek, 2019-07-28:
Just noticed with the last release device in the 'All' group went empty. When I try to add any device it shows it added in Phoscon, but does not - then a refresh of the web page shows them not in the group.

All other groups I created work as expected. Also, I tried to delete the 'All' group.. it will not delete.

salopette, 2019-07-28:
I also have the same problem

KodeCR, 2019-08-09:
The "All" group (group 0 for the API) is a build-in group that you can't add devices to or delete. As the name implies it always contains all the lights.

By last release do you mean v2.05.66? Because that contained some fixes for group 0.

holdestmade, 2019-08-09:
I have v2.05.66 and there is nothing showing in the built in All group

You can add lights to the All group but they don't show up in the Phoscon interface.
If you go back and add another light, all the ones you added previosly are now not added even though they are not shown

Anyone else's works like this ? I would assume it should auto-populate all the available lights. 

I have just created another "ALL" group and manually added all my lights

closed 2019-12-14 by stale[bot]