dresden-elektronik/deconz-rest-plugin #1716: Conbee 2 will not connect to IKEA devices

KrisBird, 2019-07-29:


Smanar, 2019-07-29:
Hello, have you tried whith "reseting" (playing with on/off) the bulb during the permit join and not before ?

KrisBird, 2019-07-29:
Yes the light bulbs have been reset by switching them on and off 6 times and I saw it blinking at the end.

Smanar, 2019-07-31:
I think there is nothing to see with your issue, but someone have a strange bug too, He need to power cycle his gateway to make pairing working > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1691

By curiosity, you haven't tried with another manufacturer ? Remote control don't pair too ?

KrisBird, 2019-07-31:
Thanks for trying to help.
I have tried now with docker, windows and a raspberry pi image from phoscon.de. I have tried pairing with both remote and brand new IKEA light bulb. I can see the conbee but it will not pair at all. Tried all different channels and different security modes.

I am in contact with dresden-electronics to see if they are able to help out.

thecem, 2019-08-02:
Did you try:

sudo systemctl stop deconz
sudo GCFFlasher_internal -r
sudo systemctl start deconz

and try to pairing in the next 5 minutes?

Sku1ly, 2019-08-02:
I had this issue and solved it but reseting the ikea lights using a Philips hue remote. Hold on and off right next to it until it flashes, goes off then on again.

Smanar, 2019-08-02:
You can try too the touchlink feature, but your bulb need to be realy close to the gateway, I can tell you how it work exactly, I have never tried it.
Go in phoscon help/old web app version then menu/setting/scan for devices.

masterpanic, 2019-11-29:
Jop it works !!!

closed 2020-04-04 by stale[bot]

thetravellor, 2021-07-24:
Im having the same problem, have just returned the stick to Amazon as faulty.