dresden-elektronik/deconz-rest-plugin #1717: Xiaomi Door contact as sensor for lighting control

salopette, 2019-07-30:
Hello, I have a door contact in the Deconz, but I can not use it for any of the lights. Neither as a switch nor as a sensor.

I wanted to set it that way when the door is open, the light goes on.
And if too light then off.

Can you do that?

Thank you

Smanar, 2019-07-30:
Hello, you can take a look here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1674
But I think it will easier to do that with your home automation application.

closed 2019-08-01 by salopette

RezzZ, 2020-03-29:
@Smanar sorry but I have to disagree with that statement.

Saying it is easier to do it with home automation might be right at this point as we cannot do this properly in Deconz but shouldn't deconz allow us to select these sensors as well in the first place?
The control via zigbee sensors should remain in deconz and should not be handled by other homeautomation software like nodered or homeassistant.

An open/close sensor is just as valid as a motion sensor to turn on/off lights. At leats for my case:
I have an attic foldable stairs and I have an open/close sensor there. If the stairs are 'open', meaning I can walk up to the attic I want my lights to stay on. If I close the attic door the lights should turn off. Same would be usable for closets etc.

If the door remains open, the light should be turned on and stay on until the door is closed again...

mgrn0, 2020-04-27:
I would love to be able to do exactly the same thing - use a door sensor to turn off a light. please?

Proche72, 2020-07-22:
Hi, is there any development on this?

SwoopX, 2020-07-23:
That is a question to be raised at the Phoscon repository, not here.

mgrn0, 2020-07-23:
@SwoopX I am confused since there was a lot discussion about this specific topic here, and not in the Phoscon repository. Can you point us in the right direction or should we open an identical issue there?  is https://github.com/dresden-elektronik/phoscon-app-beta/issues the right place?

SwoopX, 2020-07-23:
Yes, the link you mention is the right place. DE has just recently re-opened the repo to keep the topics seperated.

mgrn0, 2020-07-23:
thanks @SwoopX , opened an issue there. (@Proche72 , @RezzZ  - you might want to jump in ;)